realsenseai/librealsense #6248: Unable to removes Ir pattern from left imager

robertxwu, 2020-04-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 (sn: 950122060579)
| Firmware Version                   | 05.12.03.00
| Operating System & Version |   Windows 10 
| Kernel Version (Linux Only)    |  N/A
| Platform                                 | PC
| SDK Version                            |  v2.33.1
| Language                            |  N/A    
| Segment			|  Robot
### Issue Description

With RealSense viewer, selecting Remove Ir Pattern had no effect: failed to remove IR patterns from left RGB8 images, as shown below:
[image]

Basically, there was no difference between Default and Remove Ir Pattern presents for the left color infrared images.

All your assistance will be greatly appreciated!

Thanks, Robert

MartyG-RealSense, 2020-04-16:
I ran extensive tests with a D415 using the same settings as you.   I tested with both SDK version 2.33.1 and 2.34.0 and confirmed that the IR pattern was not removed when the pattern removal preset was used.

You can though turn the pattern off by selecting the 'Laser' drop down menu option and choosing 'Laser Auto' from the menu.

[image]

**LASER**

[image]

**LASER AUTO**

[image]

robertxwu, 2020-04-16:
@MartyG-RealSense thank you for the confirmation and the workaround.
Regarding the Laser Auto option, would you please tell me under what circumstance the laser will be turned on?
Can I do the IR pattern removal on acquired images by myself (i.e. apply some sort of filtering to the acquired images on my PC)? If so, how can I do it?

MartyG-RealSense, 2020-04-16:
The laser will be on if the "Laser Power" setting is '1' or greater.  Setting Laser Power to '0' turns it off.

IR pattern removal without the filter is possible on D415 but complicated, involving a 'Color Correction Matrix'.  Section 10c of Intel's camera tuning guide covers the subject.

https://www.intel.com/content/www/us/en/support/articles/000027833/emerging-technologies/intel-realsense-technology.html

robertxwu, 2020-04-16:
@MartyG-RealSense, I used the following code, it didn't help:
        rs2::pipeline pipe;
	rs2::config cfg;
	cfg.enable_stream(RS2_STREAM_INFRARED, RS2_FORMAT_RGB8, 30);
	cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);

	rs2::pipeline_profile profile = pipe.start(cfg);

	... // other configurations

	rs400::advanced_mode advance(profile.get_device());
	STColorCorrection colorCorrect = { 0.520508, 1.99023, 1.50684, -2, -2, -0.0820312, 1.12305, 1.01367, 1.69824, -2, 0.575195, -0.411133 };
	advance.set_color_correction(colorCorrect);

MartyG-RealSense, 2020-04-16:
I believe that color correction changes are done in a visual preset file ('json') and not a program script, as they are Advanced Mode settings.

An easy way to create a json file is to define settings in the RealSense Viewer and then use the json export button on the global at the top of the Viewer's options side-panel.

robertxwu, 2020-04-16:
@MartyG-RealSense that won't fly, as I described in the issue description.

MartyG-RealSense, 2020-04-16:
If a JSON file does not contain definitions for color correction then the color correction settings will not be changed from the default settings when the JSON is applied.

agrunnet, 2020-04-18:
This is indeed a new bug. Thanks for alerting us to this regression in the FW. We will try to fix it. 
In the mean time please try this JSON
[RemoveIR_For_FW5_12_3_100.zip](https://github.com/IntelRealSense/librealsense/files/4497059/RemoveIR_For_FW5_12_3_100.zip)
It DOES make the RGB image darker. This can be compensated for in SW by multiplying all RGB values by a constant. Alternatively you can manually increase the imager gain (although this may also affect the depth a bit).

MartyG-RealSense, 2020-04-18:
@agrunnet   Thanks so much for the information and the JSON!

closed 2020-05-17 by MartyG-RealSense