realsenseai/librealsense #6249: How to get distance on Android phone

uchengwang, 2020-04-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | android   |
| SDK Version                            |  { legacy / 2.13.0 }                          | 
| Language                            |  {android  java}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Thank you first and look forward to your reply

On Android phones, how can I get the real distance of a point in the image returned by the server? I tried the demo you gave, but in the process of moving, there is always a big difference between the real distance and the real distance in the method of realsense depthframe.getdistance (int x, int y)

在Android 手机上,  再得到服务器返回给我图片上某一点的坐标后, 我如何准确的获取这一点的真实距离? 我尝试了你们给的demo, 但是总会在移动过程中, realsense DepthFrame.getDistance(int x, int y)这个方法,获取的距离和真实距离差距很大

MartyG-RealSense, 2020-04-16:
RealSense 400 Series cameras have an error factor called 'RMS Error', where measurement error starts at near zero at the camera lens and linearly increases as an observed object moves further away from the camera.  On the D435 camera models, RMS error over distance is around 2x the RMS error of the D415 model, as shown in the curves on the chart below.

[image]

The D435 error over distance is represented by the upper orange line on the chart.

The depth accuracy of the 400 Series cameras is less than one percent of the distance from the object.  So if the camera is 1 meter from the object, the expected accuracy is between 2.5 mm to 5 mm.

You may get better accuracy if you use the optimal resolution.  On the D435 models, this is 848x480.

uchengwang, 2020-04-16:
Thank you for your reply

I get the depth distance of a specific location after the server returns the object coordinates, but there is a big gap using the above method. Do you have any relevant demo reference

uchengwang, 2020-04-16:
In addition, using the resolution of 848 * 480 on Android phones will not display anything

MartyG-RealSense, 2020-04-16:
Could you explain to me what is meant please when you say that the image is returned by the server?

uchengwang, 2020-04-16:
On an Android phone, I need to upload the obtained image information to the server, and the information returned by the server will tell me what and coordinate points are in the image. I need to obtain the corresponding depth distance based on this coordinate point. Now the obtained distance and actual The gap is very large (about 4-5 pictures will be uploaded every second, and Android phones have been constantly calculating based on the information returned by the server)

uchengwang, 2020-04-16:
Because of the problem of obtaining incorrect depth distance, I have been working for several days and have not solved it. The following work cannot be continued. I checked your official documentation, but there is very little information about Android.
If you can provide related solutions, I will be very grateful

MartyG-RealSense, 2020-04-16:
Your project sounds similar to a case I handled in March 2020.  In that setup, the camera was connected to a desktop computer that acts as a centralized server and the camera data needed to be sent to an app on the Android tablet (so that the data was interactive and not just a video of the camera data on the server).   Coordinates from user inputs onto the Android tablet would then be sent back to the server so the server could control a robot.

Does this sound like what you are trying to achieve in your own project, please?

https://github.com/IntelRealSense/librealsense/issues/5971

uchengwang, 2020-04-16:
[image]

It may be a little different. Realsense continuously transmits data to the Android phone, and the phone transmits the data to the server, and then the returned data will tell me what objects and object coordinates are in the picture. I need to constantly judge between the two objects the distance.

uchengwang, 2020-04-16:
On Android phones, can I continuously calculate the actual distance of two points through realsense? If the distance can be calculated, the Android phone will broadcast the distance, which can help people who lose their arms and blind to pick up daily necessities

MartyG-RealSense, 2020-04-16:
There have been a couple of assisted-sight projects in RealSense's history.  One of them, Sonic-Sight, was to use bat-like echo location to emit beeping sounds.

https://github.com/IntelRealSense/librealsense/issues/3629#issuecomment-478445085

In another project called Spatial Awareness Wearable (SAW), vibrating pads were attached to the body and would vibrate a specific pad if an obstacle was detected approaching that area of the body.

https://software.intel.com/en-us/articles/archived-how-to-build-the-intel-realsense-spatial-awareness-wearable

Regarding using Android to measure the distance between two points: there are such routines for C++, C# and Python, but not that I know of for Android.  It may be easier to use a simpler program to measure the distance to an observed object (instead of comparing two points) and reading the distance value out loud with speech from the app for the visually impaired user.  

The SDK has an Android example that can print the distance of the center pixel in the frame.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/java_example

uchengwang, 2020-04-17:
Thanks for your reply.
In this example, the distance is obtained when the coordinate points are known. My current requirement is to get the distance of the corresponding coordinates after the server returns the coordinates. Are there any related ideas?

kafan1986, 2020-04-17:
@uchengwang You are processing live video feed. So have you have cached depth frames? So that once the coordinate returns, you can go back to the corresponding depth frame and get the correct depth. Also, do remember about alignment of color coordinate to depth coordinate system.

MartyG-RealSense, 2020-04-17:
@kafan1986   Thanks so much for your input on this case?

@uchengwang   Does the advice of @kafan1986 help to answer your question about coordinates, please?

uchengwang, 2020-04-17:
[image]
[image]
[image]

Thank you very much for your patience!
Could you give me some suggestions about cache depth frames and alignment?
I wrote it myself, I feel the depth is not aligned with RGB.
I have been troubled by this problem for three days

uchengwang, 2020-04-17:
I am very suspicious now that the depth information may not be aligned with RGB. But I did not find the relevant method on the Android phone online. If you have relevant suggestions, I will be very grateful

kafan1986, 2020-04-17:
Couple of inputs:
1) There is no direct way of doing caching of older frames. For my use case I use opencv, and convert the color and depth frames to opencv Mat and thus allowing "Clone" operation.
2) Alignment is computationally expensive operation and is slow on phone. For you use case, you don't need to align entire image. Just the corresponding coordinate. For android there is no direct method to do this. You need to write native wrapper to access this method. https://github.com/IntelRealSense/librealsense/issues/5440#issuecomment-566593866

uchengwang, 2020-04-17:
Thank you very much for your patience. Best wishes for you

closed 2020-04-17 by uchengwang