realsenseai/librealsense #6251: NiViewer Windows DeviceOpen using default: no devices found

grafussik, 2020-04-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10)  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.33.1                           | 
| Language                            |                            | 
| Segment			|  {interactive display }                          | 

Hi!
I cant start NiViewer. Get message DeviceOpen using default: no devices found.
Sould i build wrapper on my computer or can use already builted dlls?
How OpenNI should know which device to use? 
I am very confused.
Thanks for any explain

dorodnic, 2020-04-16:
Are you following [wrappers/openni2](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openni2)?

MartyG-RealSense, 2020-04-16:
There was also the recent case @dorodnic of the recommendation to use x86 binaries.  

https://github.com/IntelRealSense/librealsense/issues/5976

grafussik, 2020-04-17:
> Are you following [wrappers/openni2](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openni2)?

Yes. I say about this wrapper.
And my NiViewer don't see any device

grafussik, 2020-04-17:
> recommendati

Thanks, It advice help me. NiViewer work, after rebuild  wrappers/openni2 in x64.
Both OpenNI2 like x32 and x64 perfect work.Thank @dorodnic

closed 2020-04-23 by dorodnic