realsenseai/librealsense #6252: Is get_units() function for depth_frame class not implemented in MATLAB?

kidhong2, 2020-04-16:
Is get_units() function for depth_frame class not implemented in MATLAB? I plan to use this to convert the depth data in terms of meters. Any other way of doing it apart from a computationally expensive loop?

* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSenseCustomerSupport, 2020-04-29:
----------------------------------------------

Thanks for your info.
Please have a try the PR https://github.com/IntelRealSense/librealsense/pull/6318 .

RealSenseSupport, 2020-05-11:
Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

kidhong2, 2020-05-12:
Hi

Thanks for your response.
I have been trying to compile the code but I can't get it to work.
I'm new to using Matlab and I'm not sure how packages work.
I did not go through the installation using source code the first time. I
just used the installer.

I tried the tutorials in the github page but to no avail.

Would you be able to help me?

Best regards
Dhong

RealSenseCustomerSupport, 2020-05-19:
----------------------------------------------

@kidhong2 Please refer to the readme document in the link below.
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/matlab

RealSenseCustomerSupport, 2020-05-26:
----------------------------------------------

@kidhong2 Any other questions about this? Looking forward to your update. thanks!

kidhong2, 2020-05-26:
Hi

still trying to run compile it.
Can't get through yet.
Sorry. Will let you know when I get it done.
I'm new to using cmake and matlab so there's a learning curve there.

Dhong

RealSenseCustomerSupport, 2020-06-03:
----------------------------------------------

@kidhong2 Is there any progress from your side? Thanks!

RealSenseCustomerSupport, 2020-06-03:
----------------------------------------------

@kidhong2 https://github.com/IntelRealSense/librealsense/pull/6318 was already merged into the latest librealsense v2.35. You can also have a try with the latest release. Thanks!

kidhong2, 2020-06-03:
Hi

Thank you very much.
This is appreciated.

Dhong

RealSenseCustomerSupport, 2020-06-22:
----------------------------------------------

@kidhong2 Any update from your side? Thanks!

RealSenseSupport, 2020-07-02:
@kidhong2 Any chance to try now? If you still can't get chance to try in these two weeks, can we close it at this point? Later if you still get the issue, please feel free to create another one. Thanks!

kidhong2, 2020-07-02:
Hi

Sorry. didn't get the chance to try it.
Please close the ticket.
I will try it sometime in the future and will create another one should I
need to.

Best regards
Dhong

RealSenseSupport, 2020-07-06:
@kidhong2  Thanks for the reply! Close this accordingly.

closed 2020-07-06 by RealSenseSupport