realsenseai/librealsense #6254: D435i Only Being Detected as USB3.0

alexander-jh, 2020-04-16:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i
| Firmware Version                   | 05.12.03.00
| Operating System                  | Windows 10
| Version                                   | 1903
| Build                                       | 18362.778
| SDK Version                            | 2
| Motherboard                          | ASUS Hero Maximus XI
| BIOS                                        | 1401

I have verified drivers and all the addressing, but for whatever reason in the Viewer the D435i USB connection is being reported as a 2.1 connection. The camera is connected to a USB 3.0 port, I have tried three different USB 3.0 cables, and gone through all hardware remedies.

I'm looking for any advice for the sake of making a connection via 3.0. Thank you.

[image]
[image]

MartyG-RealSense, 2020-04-16:
Here are some things to try first.

Please try unplugging the camera, waiting a couple of seconds and inserting the camera into the USB port quickly and firmly (not slowly).  This will reduce the chance of a USB2 mis-detection occurring.

Try the camera with the official short USB cable supplied with the camera if you are using your own choice of cable instead of official ones.

If you can access the RealSense Viewer program, try updating the camera firmware driver to the latest version by going to the More icon at the top of the Viewer's options side-panel and choosing from its menu the option to update to recommended firmware.

Also in the Viewer, you can also choose the Reset Hardware option from the More menu to restart the camera and see if it affects the USB status.

alexander-jh, 2020-04-16:
@MartyG-RealSense I tried all of those steps already unfortunately.

MartyG-RealSense, 2020-04-16:
Have you tried the camera in a different USB 3 port please?

alexander-jh, 2020-04-16:
Yes, I tried the camera on all available ports on my motherboard using the supplied short cable. Even tried a the short cable from another RS to make sure it wasn't a cable issue.

MartyG-RealSense, 2020-04-16:
In the Device Manager, please check for the drivers under the 'Cameras' section.

alexander-jh, 2020-04-16:
[image]

MartyG-RealSense, 2020-04-16:
The drivers in Cameras look fine.  Thanks!

Have you rebooted the computer and checked the USB status again after the reboot?

alexander-jh, 2020-04-16:
I have and nothing changes.

MartyG-RealSense, 2020-04-16:
Do you have a mains electricity powered USB 3 hub that you can try the camera in?

alexander-jh, 2020-04-20:
What do you mean by that?

MartyG-RealSense, 2020-04-20:
A USB hub is a box with multiple USB ports on it.  There are two types of hub called unpowered ("passive hub") and powered ("active hub").

With an unpowered hub, you connect the hub to one USB port on the computer, and so multiple USB devices can be connected to the computer whilst only using one USB port on the computer.  The hub draws its power from the computer's power supply though, which makes devices attached to the hub vulnerable to the same instabilities that can happen to devices plugged directly into the computer.

A powered hub is connected to a mains electrical plug like a wall socket instead of drawing its power from the computer.  This makes the power supplied to the USB ports on the hub more stable, which makes RealSense errors that are related to the USB port less likely to occur.

An example of unpowered versus powered is connecting a RealSense camera to a small caseless computer board such as a Raspberry Pi or Jetson Nano.  If the camera is connected directly to the board (a "passive" connection) then the camera may draw up to 2 watts of power, and a small board may have trouble providing enough power to the camera.

If a camera is attached to a mains electricity powered hub (an "active" connection) then the hub can provide 12 watts of power to devices attached to the hub, meaning that it has enough power even for more than one camera to work when attached to the hub.

alexander-jh, 2020-04-20:
I will order one asap to test if that fixes the issue. I have experienced the same miss-detection issue across several devices (specifically a Jetson Nano, through Mac OSX, and Ubuntu 18.4 on the current machine I've been troubleshooting) and only have used the on-board USB connections. I'll update the results after it arrives.

Thanks!

MartyG-RealSense, 2020-04-20:
I will look forward to hearing about your results.  Good luck!

alexander-jh, 2020-04-23:
So small improvement, there no longer is a USB overflow error with the external hub connected. It is still recognizing the camera as USB 2.1 though. Tried a several USB3 cables, two different supplied short ones, all had the same issue.

MartyG-RealSense, 2020-04-23:
If unplug-replug from the computer's USB port is not correcting the problem, please test whether unplugging the small camera end of the cable from the side of the camera and re-inserting the small end into the side of the camera helps.  

Also try inserting the small end of the cable into the camera the opposite way up, as the micro size end is a two way fit

alexander-jh, 2020-04-23:
Gave that a shot already with a few different cables. Also tried disabling non-xHCI USB devices in the UEFI with no luck either. Running short on ideas. For the sake of being thorough I fresh installed my OS to clear the registries and installed all drivers fresh.

MartyG-RealSense, 2020-04-23:
This may be a case that falls under the "Open" (unfixed) known issue documented in the firmware release notes of some cameras having problems with being properly recognised as USB 3.

[image]

It is classified elsewhere in the notes as a "No Fix" issue unfortunately.  'No Fix' means that there are no plans to fix the issue.

[image]

The RealSense team invested a significant amount of effort in investigating it.  There are some problems though where a satisfactory solution is not found no matter how much time and resources are spent on trying to resolve it.

May I please confirm though that the powered hub that you bought is a USB 3 hub, as there are also  powered USB 2 hubs available that might treat the camera as a USB 2 device.

alexander-jh, 2020-04-23:
Yep, the same one you liked earlier. It’s registered in the Device Manager as Generic SuperSpeed with appropriate drivers. 

On a side note I experienced the same issue going USB-C from host to camera as well across all three major OS’s. The primary difference originally was until I used the external supply I’d only get a video stream on my MacBook. 

At least now even though the data stream is a bit slow I can resume work. 

Is this just a silicon lottery situation where this bug is just an inconsistency in some batches?

Thanks for the help though!

MartyG-RealSense, 2020-04-23:
You're very welcome!

I think the lottery is not so much with the camera as the computers.  Not all USB ports are made equal despite being regulated by a standard.  On some machines you might find that the camera only works in some of the ports or none of them them.  But when the camera is tried on a different machine it works perfectly.  There is no way to tell in advance what results you will get.  Thankfully the number of cases in which it is impossible for the camera to work are rare.

alexander-jh, 2020-04-23:
Could it be a fair assumption at this point that the problem would be on the camera's port? I haven't seen anything past a 2.1 connection now across my desktop, three laptops, and two linux work stations using a variety of different cables.

MartyG-RealSense, 2020-04-23:
Hardware flaws and failures in RealSense cameras are very rare, though they are not impossible.  Like any electronics product, the occasional imperfect unit will leave the factory.

At this point, one of the only remaining options to try may be a 'gold reset'.  This is different from a camera reset, as it resets the camera to its factory defaults.

https://forums.intel.com/s/question/0D70P000006EFHESA4

MartyG-RealSense, 2020-04-23:
I have a recollection that another strategy for addressing this issue in the past has been to downgrade the firmware to the previous version to the currently installed one and then re-upgrade to the current version.

alexander-jh, 2020-04-23:
Tried the gold reset before I posted here unfortunately. Gave it another shot though.

MartyG-RealSense, 2020-04-23:
I just added a new comment above about firmware downgrade / upgrade.

alexander-jh, 2020-04-23:
Tried it with several older releases on both the firmware and SDK.

MartyG-RealSense, 2020-04-23:
I think that you have attempted just about everything that could be reasonably expected to rectify the problem.  I cannot think of anything else that you have not already tried.

alexander-jh, 2020-04-24:
Thank you very much for all of your help, feel confident in troubleshooting any future cameras. I submitted a ticket to Intel's customer support and they think it seems faulty, so I'm just going to exchange it.

closed 2020-04-24 by alexander-jh

MartyG-RealSense, 2020-04-24:
Thanks so much for the update.  It's great that you were able to take positives from the troubleshooting experience. Good luck!