realsenseai/librealsense #6257: Cmake error -DBUILD_PCL_EXAMPLES=true

PerceptMD, 2020-04-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (/) | 
| Operating System & Version |   { Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  (4.15.0-96-generic)       | 
| Platform                                 | PC.  |
| SDK Version                            |  {librealsense master bc64e52c9e128aa82cc9f25a14283bbe2bcf8828 }                          | 
| Language                            |  {/}                          | 
| Segment			|  {/}                          | 

### Issue Description
Previously I have successfully installed the debian librealsense packages via apt-get but now I needed to compile the PCL examples too. So I followed the "Linux manual installation guide".
But calling
`cmake ../ -DBUILD_EXAMPLES=true -DBUILD_PCL_EXAMPLES=true`
resulted in the following error

> 
> CMake Error at wrappers/pcl/pcl-color/CMakeLists.txt:13 (find_package):
>   By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
>   asked CMake to find a package configuration file provided by "OpenCV", but
>   CMake did not find one.
> 
>   Could not find a package configuration file provided by "OpenCV" with any
>   of the following names:
> 
>     OpenCVConfig.cmake
>     opencv-config.cmake
> 
>   Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
>   "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
>   provides a separate development package or SDK, be sure it has been
>   installed.
> 

This could easily be solved by running
`sudo apt-get autoremove libopencv-dev python-opencv`.
Maybe this could be updated in the installation instructions at [wrappers/pcl](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl).

MartyG-RealSense, 2020-04-17:
Thanks so much for the documentation feedback!  I have put the 'Documentation' label on this case so that it can be tracked by the team members responsible for it.

In regard to the udev rules, I hope that the link below will be helpful.

https://github.com/IntelRealSense/librealsense/issues/4350#issuecomment-516548279

PerceptMD, 2020-04-17:
`sudo rm /lib/udev/rules.d/60-librealsense2-udev-rules.rules`
deleting "60-librealsense2-udev-rules.rules" fixes the warning and the realsense-viewer still seems to works like a charm.

Thanks for your great support. It is so great to learn about the usage of the realsense SDK here on github that I want to contribute as much as I can.

MartyG-RealSense, 2020-04-17:
Awesome, I'm happy to know that your problem was fixed.  Your enthusiasm for RealSense is really appreciated.  Our community create amazing projects every day.   :)

ev-mp, 2020-04-17:
@PerceptMD , deleting files from Debians  breaks the package integrity.
You should ether delete the stand-alone file or the Debian packages altogether.
Similar to https://github.com/IntelRealSense/librealsense/issues/6153#issuecomment-605687547

PerceptMD, 2020-04-20:
Thanks for the advice.
I've already deleted the "60-librealsense2-udev-rules.rules". So i should purge the Debian package "librealsense2-udev-rules"?

closed 2026-10-23 by MartyG-RealSense