realsenseai/librealsense #6260: Announcement: Highlighting real-time reconstruction of a corridor with RealSense D435i and T265

MartyG-RealSense, 2020-04-17:
Hi everyone,

In the YouTube video link below, software developed at Intel Labs China called FAST Mapping is used in conjunction with RealSense D435i and T265 to map a long corridor in real-time.

https://www.youtube.com/watch?v=dvyha4ftj7E

Video description: "This video shows the reconstruction of a long corridor in a university building. It was done by FAST Mapping, a software package developed at Intel Labs China.  FAST Mapping takes pose-aware depth image sequences as input, and fuse all the measurements into a 3D map.  It runs in real-time on a Intel Core CPU.   The input data in this show are from the OpenLORIS-Scene datasets, recorded by a robot equipped RealSense cameras".

Further details of the project can be found at the link below.

https://lifelong-robotic-vision.github.io/dataset/scene

closed 2020-04-20 by RealSenseCustomerSupport

patrickpoirier51, 2020-05-13:
Hello @MartyG-RealSense 

Do you know if the FAST Mapping code is available on github  ?

Regards

MartyG-RealSense, 2020-05-13:
I researched your request.  One of the people who worked on this mapping project, Xuesong Shi, is a research scientist based at Intel Labs China, so he is probably nearest to Intel Labs China's "FAST Mapping" code.

https://www.intel.com/content/www/us/en/artificial-intelligence/bios/xuesong-shi.html

I could not find any open source GitHub links to the FAST Mapping system.  I recommend mailing XuesongShi to ask them about it.  I will display his email address as an image so that spam tools can't extract it from this page!

[image]