realsenseai/librealsense #6261: What data reside  inside this array ->    depth = depth_frame.get_data()

vikram09, 2020-04-18:
what data resides in depth =     depth_frame.get_data()
      
Using NP array I could see its array of width:height of Frame. But when I try to print out the data I cannot understand it. 

I can get distance at each pixel using depth_frame.get_distance(x,y) 

but what does  depth_frame.get_data() return ??? I tried to print the center pixel and its shows random numbers between 0 and 2000 etc. If depth is not 2D array of distance at each pixel , what else is it . If its array of distance , how can it be converted to Meters like the get_disatnce

MartyG-RealSense, 2020-04-18:
The pyrealsense2 documentation just says that it is a depth related function that retrieves data from the frame handle.

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.depth_frame.html

The link below describe what a frame handle is in terms of pyrealsense2.

https://github.com/IntelRealSense/librealsense/issues/5087#issuecomment-550459833

dorodnic, 2020-04-23:
Depth frames are provided in Z16 format - see [rs_sensor.h:62](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/h/rs_sensor.h#L62):
16-bit linear depth values. The depth is meters is equal to depth scale * pixel value.

vikram09, 2020-04-24:
Any idea how to use RS2_FORMAT_XYZ32F ? 
I am really lost on how to take depth information and color camera info and find the 3D co-ordinates of a point. Is there an example

vikram09, 2020-04-25:
RS2_FORMAT_XYZ32F does not work in depth or color schemes

dorodnic, 2020-04-26:
Please follow the official examples. Each includes step-by-step tutorial to help explain the relevant concepts. To get `RS2_FORMAT_XYZ32F` see [rs-pointcloud-sample](https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud#rs-pointcloud-sample)

closed 2020-05-05 by dorodnic