realsenseai/librealsense #6262: One of D435i cameras cannot stream depth anymore under realsense-viewer and python wrapper, but works under ROS. Is it a hardware malfunction or a firmware/patch problem?

szx0112, 2020-04-18:
Hi, I have multiple rs D435i cameras, and all of them work properly for a while. 

About a month ago, I notice one of the cameras fails to get depth streaming in realsense-viewer (show frame not received in 5 seconds error, color module works properly as shown in figure below). I didn't pay much attention as I majorly use them under python and ROS (and both of them works properly at that time).

Yesterday, I find I am not able to get the depth stream from python anymore. However, the camera still works properly under ROS (all other cameras work properly under python, ROS and realsense-viewer). This make me confuse if it related to the hardware malfunction (then ROS should not work either) or a firmware/patch problem (while other cameras work properly all the time)

Below I attached the info of the camera under with diagnose.

**rs-enumerate-devices** properly shows device information
```
Device info: 
    Name                          : 	Intel RealSense D435I
    Serial Number                 : 	909512072056
    Firmware Version              : 	05.12.03.00
    Recommended Firmware Version  : 	05.12.03.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb6/6-1/6-1:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B3A
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	911623050215
    Firmware Update Id            : 	911623050215

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  1280x800	@ 30Hz	   Y8
    Infrared 2	  1280x800	@ 30Hz	   Y8
    Infrared 1	  1280x800	@ 25Hz	   Y16
```

In **realsense-viewer**: stereo module starts failure as described
[image]

Use **lsusb -v**  (other cameras show similar info)
```
Bus 002 Device 002: ID 8087:8002 Intel Corp. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x8087 Intel Corp.
  idProduct          0x8002 
  bcdDevice            0.05
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           25
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
[... 145 lines cut ...]
[705772.393853] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[705772.444882] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[705772.495831] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[705772.827769] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[705777.620225] RTW: rtw_set_ps_mode(wlx0013eff3019a) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
[705777.620231] RTW: rtl8822b_set_FwPwrMode_cmd(wlx0013eff3019a): HW port id=0
[705777.620234] RTW: rtl8822b_set_FwPwrMode_cmd(wlx0013eff3019a): fw ps mode = ACTIVE, drv ps mode = 0, rlbm = 0 , smart_ps = 0, allQueueUAPSD = 0
[705783.668307] RTW: rtw_set_ps_mode(wlx0013eff3019a) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
[705783.668312] RTW: rtl8822b_set_FwPwrMode_cmd(wlx0013eff3019a): HW port id=0
[705783.668314] RTW: rtl8822b_set_FwPwrMode_cmd(wlx0013eff3019a): fw ps mode = LPS, drv ps mode = 2, rlbm = 1 , smart_ps = 2, allQueueUAPSD = 0
[705787.704337] RTW: rtw_set_ps_mode(wlx0013eff3019a) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
[705787.704342] RTW: rtl8822b_set_FwPwrMode_cmd(wlx0013eff3019a): HW port id=0
[705787.704346] RTW: rtl8822b_set_FwPwrMode_cmd(wlx0013eff3019a): fw ps mode = ACTIVE, drv ps mode = 0, rlbm = 0 , smart_ps = 0, allQueueUAPSD = 0
[705789.716372] RTW: rtw_set_ps_mode(wlx0013eff3019a) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
[705789.716378] RTW: rtl8822b_set_FwPwrMode_cmd(wlx0013eff3019a): HW port id=0
[705789.716382] RTW: rtl8822b_set_FwPwrMode_cmd(wlx0013eff3019a): fw ps mode = LPS, drv ps mode = 2, rlbm = 1 , smart_ps = 2, allQueueUAPSD = 0
[705795.768424] RTW: rtw_set_ps_mode(wlx0013eff3019a) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
[705795.768428] RTW: rtl8822b_set_FwPwrMode_cmd(wlx0013eff3019a): HW port id=0
[705795.768430] RTW: rtl8822b_set_FwPwrMode_cmd(wlx0013eff3019a): fw ps mode = ACTIVE, drv ps mode = 0, rlbm = 0 , smart_ps = 0, allQueueUAPSD = 0
[705797.780453] RTW: rtw_set_ps_mode(wlx0013eff3019a) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
```

Now I am really confused if it is a hardware malfunction or the firmware/patch problem or a combination of both. I actually tested the camera on two PC desktops, and none of them works under realsense-viewer and python, but both works under ROS.

Any comment and suggestion is really appreciate.

MartyG-RealSense, 2020-04-18:
Hi @szx0112,
 
With the RealSense Viewer, does the depth stream work the very first time that you run the Viewer after booting up the computer?  (i.e you have not yet used the Viewer and closed its window)

szx0112, 2020-04-18:
@MartyG-RealSense Thanks for your help. However, reboot does not help (tested on two PCs). The camera shows exactly same error in realsense-viewer as above mentioned.

I also found the repetitive error messages (shown below) under ROS. Normally it will disappear after camera init_reset() is enabled, but not this time.

```
 18/04 02:58:40,375 ERROR [140488396011264] (types.h:307) Cannot open '/dev/video0 Last Error: No such file or directory
 18/04 02:58:40,375 ERROR [140488396011264] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: Cannot open '/dev/video0 Last Error: No such file or directory
 18/04 02:58:40,568 ERROR [140488412796672] (types.h:307) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 18/04 02:58:40,568 ERROR [140488412796672] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 18/04 02:58:41,376 ERROR [140488396011264] (types.h:307) Cannot open '/dev/video0 Last Error: No such file or directory
 18/04 02:58:41,376 ERROR [140488396011264] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: Cannot open '/dev/video0 Last Error: No such file or directory
 18/04 02:58:41,568 ERROR [140488412796672] (types.h:307) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 18/04 02:58:41,568 ERROR [140488412796672] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 18/04 02:58:42,376 ERROR [140488396011264] (types.h:307) Cannot open '/dev/video0 Last Error: No such file or directory
 18/04 02:58:42,376 ERROR [140488396011264] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: Cannot open '/dev/video0 Last Error: No such file or directory
 18/04 02:58:42,568 ERROR [140488412796672] (types.h:307) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 18/04 02:58:42,568 ERROR [140488412796672] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 18/04 02:58:43,376 ERROR [140488396011264] (types.h:307) Cannot open '/dev/video0 Last Error: No such file or directory
 18/04 02:58:43,376 ERROR [140488396011264] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: Cannot open '/dev/video0 Last Error: No such file or directory
 18/04 02:58:43,568 ERROR [140488412796672] (types.h:307) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 18/04 02:58:43,568 ERROR [140488412796672] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 18/04 02:58:43,572 WARNING [140487038072576] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 18/04 02:58:43,572 WARNING [140487029679872] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
```
These error messages seems point to the video module problem. However, they are only shown when this single camera is plugged-in.

MartyG-RealSense, 2020-04-18:
If the problem is only occurring with one of the cameras, have you tried changing its USB cable for one of the cables from the other cameras to eliminate the possibility of a bad cable, please?

szx0112, 2020-04-18:
Confirmed not a bad cable problem (switched the cables of 3 working cameras). Same error messages in realsense-viewer. Ask me refer to [frame_metadata.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md)
[image]

MartyG-RealSense, 2020-04-18:
If it is just this one particular camera that is affected and changing the cable does not make a difference (and other cameras work in the same USB port), then it is likely related to something happening inside the camera.

The next step I would recommend trying is to perform a 'gold reset' of the camera to return it to factory default settings.  The process for doing so is described in the link below.

https://forums.intel.com/s/question/0D70P000006EFHESA4/realsense-d435i-not-properly-recognized-by-realsenseviewer?language=en_US

szx0112, 2020-04-18:
Thanks for the suggestion. While it still fails to restore to the factory. Below is a set of test results after ubstallation (ubuntu 18)
```
/usr/bin/Intel.Realsense.CustomRW -r
CustomRW for Intel RealSense D400, Version: 2.6.8.0
----------------------------------------------------------------------------------------
 /usr/bin/Intel.Realsense.CustomRW -list
CustomRW for Intel RealSense D400, Version: 2.6.8.0

error calling rs2_create_device(info_list:0x150f0c0, index:0):
    get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
----------------------------------------------------------------------------------------
/usr/bin/Intel.Realsense.CustomRW -g
CustomRW for Intel RealSense D400, Version: 2.6.8.0

Error: cannot find or open device with serial number 
----------------------------------------------------------------------------------------
/usr/bin/Intel.Realsense.CustomRW -sn 909512072056 -g
CustomRW for Intel RealSense D400, Version: 2.6.8.0

Error: cannot find or open device with serial number 909512072056
----------------------------------------------------------------------------------------

```

I notice there are several similar issues for customRW, like [D435i Firmware Recovery Failed](https://support.intelrealsense.com/hc/en-us/community/posts/360039449253) and #4708, but neither conclude a valid solution yet.

MartyG-RealSense, 2020-04-18:
May I confirm that at the time of testing the camera that has the problem, you have none of the other cameras connected to the computer, please?

szx0112, 2020-04-18:
Yep, no other cameras are connected.

MartyG-RealSense, 2020-04-18:
It is an unusual case.  The RealSense Viewer can identify the camera (though not the depth stream) and yet the gold reset cannot find the camera.  Changing the cable also has no effect.  The camera seems to be able to access RGB and motion streams but not depth.

Can the Viewer stream Infrared with this camera?

szx0112, 2020-04-18:
Well, after toggler stereo module and ir and depth frames several times. It is interesting to see that we can inconsistently receive some depth and ir frames now (sometime IR is available, sometime depth is available, sometime both and sometime neigher). However, the valid IR or depth stream only works under 640x480 based on the tests (also tested 848x480, 1280x720).

Some results:

IR good, depth gone (640x480)
[image]

IR is gone, while depth is shown (640x480).
[image]

All good (happens only once: 640x480)
[image]

Not good

848x480
[image]

640x480
[image]

MartyG-RealSense, 2020-04-18:
I recall that this effect happened to me once this week when toggling depth and IR streams on and off in the Viewer (version 2.34.0).

It is at least encouraging to see that this camera is capable of generating depth frames.  

It does not seem to be an issue with the stability of the power supplied to the USB ports, since the other cameras work fine in the same port.  If it was clearly an issue with USB power stability then plugging the camera into a mains electricity powered USB 3 hub instead of directly into the computer would have a good chance of reducing / removing errors.  

There is certainly no harm in trying this particular camera with a powered USB 3 hub if that is an option available to you. 

What you could try first though is reducing the Viewer's "Laser Power" setting to a low level such as '1' (setting it to 0 would turn the projector off).

szx0112, 2020-04-18:
@MartyG-RealSense  At least we get some images. All my 6 D435i cameras use exactly same usb port, firmware and configuration steps during testing and streaming, and this is the only camera got those issues.

Powered USB hub has tried, results in same error. 

Laser power increased (upto 300) and decreased (30 is the lowest I can get except 0 without the 
`Device or resource Busy` message pops up). However, it all depends on whether the depth/IR is streaming when the Stereo Module is enabled in the APP. If it works from the start, everything looks fine. Otherwise, only blank screen with `No Frames Received Warning`. Noted, up to now, the successful cases are only performed at 640x480, other any resolutions won't work either.

MartyG-RealSense, 2020-04-18:
What is your CPU usage like during streaming?  If it is maxing out at 100% when using resolutions above 640x480 then that could be an indicator of why streaming is unreliable.

Also, what is your average ASIC temperature in the Viewer side-panel when streaming, please?

szx0112, 2020-04-18:
Please see the snapshot. Temperature looks good, while cpu usage of single thread seems really high. 
[image]

In comparison, I put a good camera result below (in 640x480)
[image]

In 1280x720
[image]

Just by observation, the problem camera indeed take a little high CPU usage than the good camera. And resolution seems not a significant issue for my case.

MartyG-RealSense, 2020-04-18:
Thanks for the images. The official recommended maximum operating temperature for the 400 Series cameras is 35 degrees C, so your temperature readings are fine.

I see in an earlier message that you tested with Ubuntu 18, so I guess that you build the SDK with Camera?  If you have an Nvidia GPU in your PC then you could offload some processing work from the CPU onto the GPU by building the SDK with CUDA support.

https://github.com/IntelRealSense/librealsense/issues/4905#issuecomment-533854888

szx0112, 2020-04-19:
Great suggestion! I will give a try for the GPU enabled version, I indeed see the rendered frames drop in the APP when 4 or more cameras streamed together.

closed 2020-05-17 by MartyG-RealSense