realsenseai/librealsense #6264: about get the imu and stereo image separately at the same time

jyzhong, 2020-04-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi，i can not get the imu and stereo image at the same time, some imu are dropped. Can you update a sample or give me some clue?

MartyG-RealSense, 2020-04-18:
If you activate the Stereo and Motion modules at the same time, are you getting frame drops on the IMU streaming, please?

What computer device are you using the camera with please?   (PC / Mac / Raspberry Pi / Jetson / Android etc)

And which operating system?  (Windows / MacOS / Linux etc)

jyzhong, 2020-04-18:
> If you activate the Stereo and Motion modules at the same time, are you getting frame drops on the IMU streaming, please?
> 
> Are you using a T265 or a D435i camera?
> 
> What computer device are you using the camera with please? (PC / Mac / Raspberry Pi / Jetson / Android etc)
> 
> And which operating system? (Windows / MacOS / Linux etc)

D435i, PC, Ubuntu. Now,  i have not the dropping imu or image. Now i wanna get the same frequency of gyro and acc.

MartyG-RealSense, 2020-04-18:
Thank you very much.  You would like to align the IMU data to depth data, please?

jyzhong, 2020-04-18:
> Thank you very much. You would like to align the IMU data to depth data, please?
I wanna get the Infrared image and the IMU data.

MartyG-RealSense, 2020-04-18:
Ok, thanks.  You are trying to do this with programming, like C++, and not with the RealSense viewer program?

jyzhong, 2020-04-18:
> Ok, thanks. You are trying to do this with programming, like C++, and not with the RealSense viewer program?
en

MartyG-RealSense, 2020-04-18:
I do apologise.   What is 'en' please?

jyzhong, 2020-04-18:
Sorry. I mean “Yes”.

发自我的iPhone

MartyG-RealSense, 2020-04-18:
Ok, thank you very much.  So you would like a C++ sample program that shows the infrared image and the IMU data of the D435i.  I will search for information for you.  Thank you for your patience.

jyzhong, 2020-04-18:
@MartyG-RealSense 
Sorry. I mean "Yes". I wanna the same frequency of gyro and acc as the version of ROS do. If i should set some paras so the imu and infrared image is synchronous ?

MartyG-RealSense, 2020-04-18:
Thank you.   I hope the very detailed discussion in the link below about aligning IMU measurements to IR frames will be helpful.

https://github.com/IntelRealSense/librealsense/issues/3942

jyzhong, 2020-04-20:
> Thank you. I hope the very detailed discussion in the link below about aligning IMU measurements to IR frames will be helpful.
> 
> #3942

Hi, after i run sample "motion". I found that the of frequency of gyro and acceleration is unstable. For example, i set 400hz for gyro, i get many drop frame and the interval of data isn't 2.5ms, maybe 3ms or 2ms. So do the acceleration. I wish get your help. Is this the right data ?  Is my sdk wrong?

MartyG-RealSense, 2020-04-20:
I thought about your case very carefully.  As you are trying to use IMU and infrared data together, the discussion in the link below may help you to program ways to generate a better output.

https://github.com/IntelRealSense/librealsense/issues/4386

If you are testing the IMU on its own in the "Motion Module" controls of the RealSense Viewer at 400hz and getting a time period of worse than 2.5 ms and dropped frames ... Doronhi the RealSense ROS developer has suggested checking whether your PC is overloaded at the time (for example, whether your CPU usage is close to or at 100%).

jyzhong, 2020-04-21:
@MartyG-RealSense  Thanks. The  CPU usage is bigger than 100%. Actually i just run the sample. How can i get the right imu data without dropped frame? BTW, when i install the SDK,  "./scripts/patch-realsense-ubuntu-lts.sh " is failed because of "fatal: index-pack failed", but i still finished installation.

MartyG-RealSense, 2020-04-21:
To avoid the need for patching, you could try the 'bypass' method of installation that is confirmed to work with D435i.  This method is also not reliant on Linux versions or kernel versions, so it is a good choice to try if you have problems with the usual Linux installation process.  The bypass installation needs an internet connection to work.

https://support.intelrealsense.com/hc/en-us/community/posts/360037657234/comments/360009799533

jyzhong, 2020-04-21:
@MartyG-RealSense thanks! i install the source code using the "bypass". Open the "realsense-viewer", i found i can get the "hardware time". i click on the button "global time", but it's still have "hardware time". When i use the "dkms"

MartyG-RealSense, 2020-04-21:
I do not have an Ubuntu setup to test with as I am a Windows user.  I ran some tests with 'Global Time' togled on and off in the RealSense Viewer on my Windows setup, at the bottom of the "Controls" section of the stereo module's side-panel options.  

When 'Global Time Enabled' was enabled, this was reflected in the stream information by showing it to be using 'Global Time'.

[image]

When the 'Global Time Enabled' option was disabled, the Viewer automatically changed 'Global Time' to 'Hardware Clock'.  Toggling 'Global Time Enabled' back on during streaming automatically switched the information back from 'Hardware Clock' to 'Global Time'. 

[image]

Are you unable to get Global Time on your own Viewer on Ubuntu, please?

jyzhong, 2020-04-21:
@MartyG-RealSense yes.  when i try "cmake ../ -DBUILD_EXAMPLES=true".  The acceleration get the two types. But the gyroscope　only have the "system" and mention "Per-frame metadata is not enable at the OS level, please follow the installation guide for details". BTW, i install the SDK according to the "DKMS", the realsense-viewer  work well. Actually, Maybe the issue trouble me is the that when i use the "callback" rather than "wait_for_stream", i can get the IR frame and imu data, but just like i say, the gyroscope have lots of dropped frame. Maybe the next version will give a sample about "align the IR and imu", or "color and imu". Thanks your for your nice reply.

MartyG-RealSense, 2020-04-21:
My understanding is that the "backend" bypass method installs support for metadata on Linux, but when using the normal installation method metadata support for Linux has to be patched in (metadata is supported by default on Windows).

https://github.com/IntelRealSense/librealsense/issues/3686#issuecomment-515775903

If you would like to make a request for the sample programs that you described, I recommend creating a new issue just for that request, with the words 'Feature Request' in the title.  I can then label the issue as a feature request so that the RealSense developer team can track it.   Thanks!

jyzhong, 2020-04-21:
Thanks.

发自我的iPhone

 

 
#3686 (comment)

closed 2020-05-05 by dorodnic