realsenseai/librealsense #6265: Cubemos Skeleton Tracking (can't build samples)

kompaso, 2020-04-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435} |  
| Operating System & Version |  { Ubuntu 18} |  
| Platform                                 | {PC }|
| SDK Version                            | {cubemos-SkeletonTracking_2.3.0} | 
| Language                            |  {C++ }| 
| Segment			|  {instalation }                          | 

### Issue Description
Good afternoon. 
I am trying to build an example from Cubemos Skeleton Tracking SDK (cpp-realsense). But when I am trying to build it with cmake, I got this mistake:
``` /usr/bin/ld: cannot find -lcubemos_skeleton_tracking ````
I have modified my LD_LIBRARY_PATH to libcubemos_skeleton_tracking.so, but it didn't help and I am not sure that it was a good idea. 
Do I understand correctly that my computer does not see the installed library? Or did I install something wrong? There were no problems or warnings during installation.
I really hope for your answer.

With best wishes

MartyG-RealSense, 2020-04-18:
I have copied the information from the Cubemos Skeleton Tracking SDK's PDF manual for building the samples on Linux.  Please compare the steps below with the installation steps that you performed.
.......................................

**Building SkeletonTracking SDK Samples with CMake on Linux**

During the Linux installation the following environment variable is created:
$CUBEMOS_SKEL_SDK | /opt/cubemos/skeleton_tracking 

**$CUBEMOS_SKEL_SDK/samples** contains the source code of the samples.   
 
1. Install CMake and build tools with 
**sudo apt-get install cmake build-essential** 
 
2. Create a directory where the built samples should be stored  
**mkdir /path/buildDir**
**cd /path/buildDir** 
 
3. Run cmake and build the samples 
**cmake $CUBEMOS_SKEL_SDK/samples**  
**make**

kompaso, 2020-04-19:
Thanks for your quick answer! The problem has been resolved.
The problem was not in this, but in my system settings.

closed 2020-04-19 by kompaso

MartyG-RealSense, 2020-04-19:
Thanks so much for the update.   I'm pleased you were able to find a solution.  Good luck with Cubemos!