realsenseai/librealsense #6266: Doc fix causing Travis build failure

himanshugarg, 2020-04-19:
I fixed a few typos in doc/installation.md via #6263 and the build [failed](https://travis-ci.org/github/IntelRealSense/librealsense/builds/676537275) during live-test.exe. If that's not due to the fix, will the PR still be picked up for review?

MartyG-RealSense, 2020-04-19:
Whilst I do not have experience in the workings of the Travis-CL test process,. It seems unlikely that changes to plain-text instructions would cause a build failure.

I had a look at the full log of the test process.

https://travis-ci.org/github/IntelRealSense/librealsense/jobs/676537276

The end of the log highlights the following line in red:

**The command "./live-test.exe -d yes -i [software-device]" exited with 1.**

This command is checked at line 1502 of the log.

https://travis-ci.org/github/IntelRealSense/librealsense/jobs/676537276#L1502

The build process seems to have passed when tested on most operating systems and languages (Windows, Linux, Mac, Android, C#, C++) but finally fails at 21 minutes into the test process when checked on Windows 7.

https://travis-ci.org/github/IntelRealSense/librealsense/jobs/676537276

himanshugarg, 2020-04-20:
Thanks, Yes the build failure is unrelated. The PR, after review, will be considered for the next release.

closed 2020-04-20 by himanshugarg