realsenseai/librealsense #6268: How to get the RGB pixel value and depth value at the pixel point (x,y) after the depth map and RGB map are aligned?

Steven-ZY, 2020-04-19:
How to get the RGB pixel value and depth value at the pixel point (x,y) after the depth map and RGB map are aligned? You know, the sample provided by the SDK  just shows us the rendering map, not data after aligning. By the way, I am new to use RGB-D, to be exact, the D435i , I want to know how to combine it with other algorithms such as object detection and pose estimation? Even I can't understand the samples in the SDK? How can I start it? Thanks in advance.

MartyG-RealSense, 2020-04-19:
Intel Python tutorial in the link below is a good introduction to aligning depth and RGB data from a bag file, identifying objects from the data and calculating distance from the camera to that object.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb

As mentioned in your other recent case about pose estimation ( https://github.com/IntelRealSense/librealsense/issues/6255 ), if you do not want to use Aruco image tags then some kind of AI solution will likely be required if you want to calculate the pose of objects.

I did some research into object detection and pose systems in the Python language that might be accessible for someone new to such programs.  the link below looks as though it would meet your need for both object detection and pose calcuation:

https://github.com/microsoft/singleshotpose

Steven-ZY, 2020-04-19:
Dear Sir，you are really a nice man. Thank you so much for your attention to my questions! I just opened the link you suggested, maybe it's suitable for me, I'll have a try. But I’m concentrating on using C++ now. If you have some idea about C++, please let me know, thank you.

MartyG-RealSense, 2020-04-19:
You are very welcome.  :)

The majority of such programs that I have seen tend to be in the Python language.  Of the C++ Deep Neural Network (DNN) object detection programs on the RealSense SDK's examples page, they usually also require use of extra software such as OpenCV or OpenVINO alongside C++.  Or else they are designed to detect human body / face poses instead of objects.

Nvidia have a program for object detection and pose estimation.  The source code for that seems to be in the **.cs** format, suggesting that it is C#.

https://github.com/NVIDIA/Dataset_Synthesizer

If you are willing to consider ROS, I found this object detection and pose estimation program for robot arms whose labelling claims that it is C++:

https://github.com/avinashsen707/AUBOi5-D435-ROS-DOPE

[image]

ev-mp, 2020-04-20:
@Steven-ZY , the [`rs-align`](https://github.com/IntelRealSense/librealsense/tree/master/examples/align) c++  demo shows how to generate Depth to RGB and RGB to Depth alignment. Once established, the (i,j) coordinate  provides a direct mapping between Depth and Color frames.

The following phrase is not clear, can you elaborate ?:
> You know, the sample provided by the SDK just shows us the rendering map, not data after aligning.

ev-mp, 2020-04-27:
@Steven-ZY, do you need more info here?
Please update

closed 2020-05-17 by MartyG-RealSense