realsenseai/librealsense #6270: Error building NodeJS wrapper on macOS : no such file or directory: librealsense2.dylib

CaseyPYZ, 2020-04-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version | MacOS Catalina 10.15.4  | 
| SDK Version                            | GitHub                          | 

### Issue Description

Hi!

I am new to realsense and was trying to build the realsense library from sourse following the [installation guidelines here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md).
I succeeded the first time when I did exactly what installation_osx.md said, ( just wanted to have the examples in librealsense/examples working (such as pointcloud)), but failed later when I tried to build along with the nodejs wrapper.

When I did:

```
cd wrappers/nodejs
npm install
```

or:

```
cmake -DBUILD_NODEJS_BINDINGS=1 -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
make -j2
```

the following message comes up:

> Linking CXX executable rs-capture
> clang: error: no such file or directory: '/Users/user/Documents/librealsense/wrappers/nodejs/../../build/Release/librealsense2.dylib'
> make[3]: *** [Release/node_librealsense.node] Error 1
> gyp ERR! build error 
> gyp ERR! stack Error: `make` failed with exit code: 2
> gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
> gyp ERR! stack     at ChildProcess.emit (events.js:198:13)
> gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
> gyp ERR! System Darwin 19.4.0
> gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
> gyp ERR! cwd /Users/user/Documents/librealsense/wrappers/nodejs
> gyp ERR! node -v v10.16.3
> gyp ERR! node-gyp -v v3.8.0
> gyp ERR! not ok 
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! node-librealsense@0.2130.0 install: `node-gyp rebuild`
> npm ERR! Exit status 1
> npm ERR! 
> npm ERR! Failed at the node-librealsense@0.2130.0 install script.
> npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     /Users/user/.npm/_logs/2020-04-19T18_57_35_423Z-debug.log
> make[2]: *** [../wrappers/nodejs/build/Release/node_librealsense.node] Error 1
> make[1]: *** [wrappers/nodejs/CMakeFiles/BUILD_NODE_ADDON.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....

I also tried installing the [node-librealsense module](https://www.npmjs.com/package/node-librealsense) directly from a nodejs project following its instructions, but also failed:

> .
> .
> .
> 
> PhaseScriptExecution CMake\ Rules build/librealsense2.build/Release/ALL_BUILD.build/Script-FF24ED55DD574003A237ED1E.sh
>     cd /Users/user/Desktop/_...path-to-my-project..._/webrtc-work/node_modules/node-librealsense/librealsense
>     /bin/sh -c /Users/user/Desktop/_...path-to-my-project..._/webrtc-work/node_modules/node-librealsense/librealsense/build/librealsense2.build/Release/ALL_BUILD.build/Script-FF24ED55DD574003A237ED1E.sh
> 
> echo ""
> 
> echo Build\ all\ projects
> 
> Build all projects
> 
> ** BUILD SUCCEEDED **
> 
> > node-librealsense@0.2130.0 install /Users/user/Desktop/_...path-to-my-project..._/webrtc-work/node_modules/node-librealsense
> > node-gyp rebuild
> 
> .
> .
> .
>                              ^
> fatal error: too many errors emitted, stopping now [-ferror-limit=]
> 8 warnings and 20 errors generated.
> make: *** [Release/obj.target/node_librealsense/src/addon.o] Error 1
> gyp ERR! build error 
> gyp ERR! stack Error: `make` failed with exit code: 2
> gyp ERR! stack     at ChildProcess.onExit (/Users/user/.nvm/versions/node/v12.16.1/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
> gyp ERR! stack     at ChildProcess.emit (events.js:311:20)
> gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
> gyp ERR! System Darwin 19.4.0
> gyp ERR! command "/Users/user/.nvm/versions/node/v12.16.1/bin/node" "/Users/user/.nvm/versions/node/v12.16.1/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
> gyp ERR! cwd /Users/user/Desktop/_...path-to-my-project..._/webrtc-work/node_modules/node-librealsense
> gyp ERR! node -v v12.16.1
> gyp ERR! node-gyp -v v5.0.5
> gyp ERR! not ok 
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! node-librealsense@0.2130.0 install: `node-gyp rebuild`
> npm ERR! Exit status 1
> npm ERR! 
> npm ERR! Failed at the node-librealsense@0.2130.0 install script.
> npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     /Users/user/.npm/_logs/2020-04-19T19_21_36_877Z-debug.log
> 

Sorry if there's anything very obvious that I'm doing wrong here, still teaching myself :) Please let me know if there's any other info I need to look at / provide.
I appreciate it greatly if someone could point me to the right direction!

MartyG-RealSense, 2020-04-19:
Usually, a CMake build instruction has two full-stops in front of it, like this:

cmake **..** -DBUILD_NODEJS_BINDINGS=1 -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
make -j2

Also, the CMake instruction is usually run after changing to the **build** folder of Librealsense after making that folder.  To quote point 4 of the MacOS installation instructions:

mkdir build && cd build
sudo xcode-select --reset

CaseyPYZ, 2020-04-20:
> Usually, a CMake build instruction has two full-stops in front of it, like this:
>
> cmake **..** -DBUILD_NODEJS_BINDINGS=1 -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
> make -j2 [...]

Thanks for the reply!

I did:
```
mkdir build && cd build
sudo xcode-select --reset
```
in my previous attempts.

I tried (in the build directory):

```
cmake .. -DBUILD_NODEJS_BINDINGS=1 -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
make -j2
```

And the same error came up:

> clang: error: no such file or directory: '/Users/user/Documents/librealsense/wrappers/nodejs/../../build/Release/librealsense2.dylib'
> make[3]: *** [Release/node_librealsense.node] Error 1

Then I looked again in the directorys and realized the library "librealsense2.dylib" is located in librealsense/build/ and there was no "Release" directory

My file structure looked like this:
[image]

CaseyPYZ, 2020-04-20:
> Usually, a CMake build instruction has two full-stops in front of it, like this:
>
> cmake **..** -DBUILD_NODEJS_BINDINGS=1 -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
> make -j2 [...]

Thanks for the reply!

I did:

mkdir build && cd build
sudo xcode-select --reset
in my previous attempts.

I tried (in the build directory):

cmake .. -DBUILD_NODEJS_BINDINGS=1 -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
make -j2
And the same error came up:

clang: error: no such file or directory: '/Users/user/Documents/librealsense/wrappers/nodejs/../../build/Release/librealsense2.dylib'
make[3]: *** [Release/node_librealsense.node] Error 1

Then I looked again in the directorys and realized the library "librealsense2.dylib" is located in librealsense/build/ and there was no "Release" directory

My file structure looked like this:
[image]

CaseyPYZ, 2020-04-20:
> Usually, a CMake build instruction has two full-stops in front of it, like this:
>
> cmake **..** -DBUILD_NODEJS_BINDINGS=1 -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
> make -j2 [...]

Thanks for the reply!

I did:

mkdir build && cd build
sudo xcode-select --reset
in my previous attempts.

I tried (in the build directory):

cmake .. -DBUILD_NODEJS_BINDINGS=1 -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
make -j2
And the same error came up:

clang: error: no such file or directory: '/Users/user/Documents/librealsense/wrappers/nodejs/../../build/Release/librealsense2.dylib'
make[3]: *** [Release/node_librealsense.node] Error 1

Then I looked again in the directorys and realized the library "librealsense2.dylib" is located in librealsense/build/ and there was no "Release" directory

My file structure looked like this:
https://drive.google.com/file/d/1dr-Ztwpd_jd6O9DbZ0c1S4FWIjfn28jp/view?usp=sharing

CaseyPYZ, 2020-04-20:
> Usually, a CMake build instruction has two full-stops in front of it, like this:
>
> cmake **..** -DBUILD_NODEJS_BINDINGS=1 -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
> make -j2 [...]

I'm wondering why that would happen (what I did wrong)?

Just out of curiosity I created a Release directory & copied a librealsense2.dylib into it, then did:
```
cmake .. -DBUILD_NODEJS_BINDINGS=1 -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
make -j2
```
as you suggested. The build finished without errors showing.

However when I tried to run the wrappers/nodejs/examples, this error showed up:

> /Users/user/Documents/librealsense/wrappers/nodejs/node_modules/bindings/bindings.js:83
>         throw e
>         ^
> 
> Error: dlopen(/Users/user/Documents/librealsense/wrappers/nodejs/build/Release/node_librealsense.node, 1): Library not loaded: @rpath/librealsense2.2.34.dylib
>   Referenced from: /Users/user/Documents/librealsense/wrappers/nodejs/build/Release/node_librealsense.node
>   Reason: image not found
>     at Object.Module._extensions..node (internal/modules/cjs/loader.js:807:18)
>     at Module.load (internal/modules/cjs/loader.js:653:32)
>     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
>     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
>     at Module.require (internal/modules/cjs/loader.js:692:17)
>     at require (internal/modules/cjs/helpers.js:25:18)
>     at bindings (/Users/user/Documents/librealsense/wrappers/nodejs/node_modules/bindings/bindings.js:76:44)
>     at Object.<anonymous> (/Users/user/Documents/librealsense/wrappers/nodejs/index.js:7:32)
>     at Module._compile (internal/modules/cjs/loader.js:778:30)
>     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)

neilyoung, 2020-04-20:
Out of curiosity (never tried to compile the node bindings): But did you make `sudo make install` finally?

MartyG-RealSense, 2020-04-20:
@CaseyPYZ  when you say that you "tried to run the wrappers/nodejs/examples", do you mean that you followed the instructions for running the NodeJS examples by:

1.  Changing directory to the 'wrappers/nodejs/examples' directory with this instruction:

**cd wrappers/nodejs/examples**

2.  And then used the installation commands below:

**npm install**
**node nodejs-capture.js**

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs#2-run-examples

CaseyPYZ, 2020-04-21:
@MartyG-RealSense Yes.

MartyG-RealSense, 2020-04-21:
Somebody else experienced this "Library not loaded: @rpath" error on MacOS in Ocrtober 2018.  

https://github.com/IntelRealSense/librealsense/issues/2471#issuecomment-426637368

Within the same discussion, a member of the RealSense GitHub team called @RealSense-Customer-Engineering  offered some advice on linking.

https://github.com/IntelRealSense/librealsense/issues/2471#issuecomment-432160378

CaseyPYZ, 2020-04-21:
@neilyoung 
No, where would this step fit into?

What I did was:
```
mkdir build && cd build
sudo xcode-select --reset
cmake .. -DBUILD_NODEJS_BINDINGS=1 -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
make -j2
```

Alex-OPTIM, 2020-05-15:
I'm having the same issue, has it been resolved?

MartyG-RealSense, 2020-05-15:
@Alex-OPTIM   I did find another previous case of NodeJS having problems on MacOS in the past.  It is not exactly the same problem, but I will provide the link below in case it gives you some ideas.

https://github.com/IntelRealSense/librealsense/issues/1592

In that particular case, the RealSense user eventually found that upgrading their camera firmware made the problem disappear.

Alex-OPTIM, 2020-05-15:
Thanks for directions @MartyG-RealSense but it didn't help.

Basically, from `npm i node-librealsense`  I have `Unrecoverable! error native function rs2_pipeline_start: No device connected`. And when I build it from source it fails on `librealsense2.dylib not found`. I fix it by creating a `Release`  folder in `../build` and copy the file there. Now it builds as per [Build with Cmake](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs#build-with-cmake) but when I run

    cd wrappers/nodejs/examples
    npm install
    node nodejs-capture.js

I get `segmentation fault  node nodejs-capture.js`. 

No luck at all.

PS. I downgraded node to v8 and I'm strictly following **macOS Installation** instruction.

MartyG-RealSense, 2020-05-15:
@Alex-OPTIM  Apologies for the delay in responding further.

When carefully researching your case further, I found another NodeJS case from February 2020 where that NodeJS user also experienced the  **"Unrecoverable!  error native function rs2_pipeline_start: No device connected"** error.

https://github.com/IntelRealSense/librealsense/issues/5856

Which RealSense camera model are you using please?  If it is the D435i, it does not yet have support for MacOS.  This is taken from the version details for the currently most recent SDK version (**2.34.0**) on the SDK 'Releases' page

[image]

Alex-OPTIM, 2020-05-16:
It's ok @MartyG-RealSense, thanks for your help. Yes, it is D435i and I feel so dumb, how did I miss it?

Just tried to build from source on my Windows machine and **it works**!

However npm package isn't working and fails with _"Unrecoverable! error native function rs2_pipeline_start: No device connected"_. So I had to copy and replace `nodejs\build\Release` folder from the build to the npm package `node-librealseanse\build\Release`. Now I can use it in my code. Yay! :) [but another challenge is awaiting to create `typings.d.ts`]

If anyone interested I'm on Windows 10 (v1903), NodeJS v10.20.1, `npm i -g windows-build-tools`, Visual Studio Community 2019, Python v2.7.15 and CMake v3.16.0.
And my steps in Command Prompt below:
- `git clone https://github.com/IntelRealSense/librealsense.git`
- `cd librealsense && mkdir build && cd build`
- `cmake .. -DBUILD_SHARED_LIBS=ON`
- `msbuild librealsense2.sln`
- `cd ../wrappers/nodejs `
- `npm install`
- `cd examples `
- `npm install`
- `node nodejs-capture.js`

Thanks RealSense team for your help.

MartyG-RealSense, 2020-05-16:
@Alex-OPTIM   Don't worry about missing the information about D435i.  An old saying goes "Pobody's nerfect".   :)

Great to hear that you had success with the build process on Windows.  Thanks so much for posting your procedure so that others can benefit from the knowledge.  Good luck!

MartyG-RealSense, 2020-05-24:
Hi @Alex-OPTIM    Do you require further assistance on this case, or can it be closed?   Thanks!

Alex-OPTIM, 2020-05-24:
Hi @MartyG-RealSense, it's all good for me, you were a great help.

Also I didn't open the issue, @CaseyPYZ are ok to close it?

MartyG-RealSense, 2020-05-24:
@Alex-OPTIM    Thanks very much!

@CaseyPYZ  Do you require further assistance on this case, or can it be closed? Thanks!

closed 2020-05-30 by MartyG-RealSense