realsenseai/librealsense #6271: How to get frame data from frame_queue in python ?

geaxgx, 2020-04-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 realsen | 
| Firmware Version                   | 5.12.03.00 | 
| Operating System & Version |   Ubuntu 18.4  | 
| Kernel Version (Linux Only)    | 4.15.0-50-generic                                      | 
| Platform                                 | PC  |
| SDK Version                            |  pyrealsense2-2.34.0.1470                         | 
| Language                            |  python                      | 

### Issue Description
In my application, I would like to minimize the lag as much as I can : I don't mind losing frames, but when I am ready to get frame, I want to get the most recent frame.
From my understanding (please correct me if I am wrong), the way to do it is to use a frame_queue with a capacity of 1.

I have written a short piece of code to reproduce the problem I have :

```
import pyrealsense2 as rs
import numpy as np

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 360, rs.format.z16, 30)
queue = rs.frame_queue(1)
pipeline.start(config, queue)

for _ in range(2):
    frame = queue.wait_for_frame()
    print("Frame #:",frame.get_frame_number())
    print("Frame size:", frame.get_data_size())
    print("Frame profile:", frame.profile)
    depth = np.asanyarray(frame.get_data()) 
    print("Depth shape", depth.shape)
    print("Depth", depth)    
pipeline.stop()
```
And here is the output I get :
```
Frame #: 0
Frame size: 460800
Frame profile: <pyrealsense2.video_stream_profile: 1(0) 640x360 @ 30fps 1>
Depth shape (0,)
Depth []
Frame #: 1
Frame size: 460800
Frame profile: <pyrealsense2.video_stream_profile: 1(0) 640x360 @ 30fps 1>
Depth shape (0,)
Depth []
```
You can see that I am unable to get the data of the frames with the get_data() call.
What am I missing ?  Thank you !

ev-mp, 2020-04-20:
@geaxgx , you need to cast depth_frame with `as_frame()`
```python
depth = frames.get_depth_frame()
depth_data = depth.as_frame().get_data()
np_image = np.asanyarray(depth_data)
```
See [Python Documentation](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#numpy-integration)

geaxgx, 2020-04-20:
@ev-mp Thanks for your reply, but it doesn't work for me.
I have replaced in my code above :
`depth = np.asanyarray(frame.get_data())`
by:
`depth = np.asanyarray(frame.as_frame().get_data()) `
But I still get exactly the same output.
I have also tried `as_depth_frame()` instead of `as_frame()`. This time I get an error message:
```
Frame #: 0
Frame size: 460800
Frame profile: <pyrealsense2.video_stream_profile: 1(0) 640x360 @ 30fps 1>
Traceback (most recent call last):
  File "bug.py", line 16, in <module>
    depth = np.asanyarray(frame.as_depth_frame().get_data()) 
RuntimeError: null pointer passed for argument "frame_ref"
```

ev-mp, 2020-04-20:
@geaxgx , the caveat is that the return value `pipe.wait_for_frames()` (plural) and `queue.wait_for_frame()` (singlular)  are not identical - they are `frameset` and `frame` accordingly.
So you need to go an extra-mile to convert the result into depth frame:

>frame = queue.wait_for_frame()
**depth_frame = frame.as_frameset().get_depth_frame()** # Extract the depth frame from queue
depth_pixels = np.asanyarray(depth_frame.get_data()) 
print("Depth shape", depth_pixels.shape)

geaxgx, 2020-04-20:
@ev-mp Thank you very much ! It works with your instructions. I never would have guessed by myself. Is it documented somewhere ? Or in some example code ?

closed 2020-04-20 by geaxgx