realsenseai/librealsense #6272: Unable to re-start the rs2 pipeline.

piotr-tanski, 2020-04-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Intel RealSense T265                                                       | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version |   Win 10 Pro | 
| Platform                                 |  PC                                                         |
| SDK Version                            |  2.33.1                                                   | 
| Language                            |  C++                                                         | 
| Segment			|  Other | 

Hi.
I've got a following problem with Realsense T265 device: after the call to rs2::pipeline::close() method, the next call to rs2::pipeline::start() hangs forever. The only way to make the program work again is to unplug the usb cable of t265 device and plug it again. The issue is reproducible every time.
To show that, I've prepared a simple C++ program which opens and closes the rs2::pipeline multiple times.

Could you please advise, how to properly re-start the device without closing the app or unplugging the usb?

```
int main(int, char**)
{
    constexpr std::chrono::seconds timeout{10};
    while (true)
    {
        // Start
        rs2::config config{};
        config.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
        rs2::pipeline pipeline{};
        pipeline.start(config);

       std::this_thread::sleep_for(timeout);

        // Stop
        pipeline.stop();
    }
    return 0;
}

Alnilam1305, 2020-04-24:
For me your example seems to work, but it is unstable. It differs on every try, and hangs up after 5 to 20 rounds in the endless loop.

RealSenseSupport, 2021-03-17:
The PR 8561 could be used in order to see if it fixes your issue.  Incorporate the PR code in your build and give it a try.