realsenseai/librealsense #6274: piecewise linear complex && piecewise smooth complex

thambimani, 2020-04-20:
what is the difference between the piecewise linear complex and piecewise-smooth complex
I saw these words in Delaunay triangulation 
knowing persons pls give clarity about this
thanks in advance

MartyG-RealSense, 2020-04-20:
Wikipedia may be a useful reference for you.  If you search for **piecewise smooth** then it redirects to **piecewise**.  so I assume that this is the entry explaining Piecewise Smooth:

https://en.wikipedia.org/wiki/Piecewise

**Piecewise Linear** has its own separate entry:

https://en.wikipedia.org/wiki/Piecewise_linear_function

thambimani, 2020-04-20:
> Wikipedia may be a useful reference for you. If you search for **piecewise smooth** then it redirects to **Piecewise**. so I assume that this is the entry explaining Piecewise Smooth: [...]

your reply is terms of mathematical but I expected  terms of complex(Delaunay structure )

thambimani, 2020-04-20:
I expected like brief about these things

MartyG-RealSense, 2020-04-20:
Sometimes, maths cannot be explained in a single paragraph, unfortunately!

thambimani, 2020-04-21:
> Sometimes, maths cannot be explained in a single paragraph, unfortunately!

that's fine can you explain why want piecewise linear complex and piecewise-smooth complex.

I know PSC is an output structure.
what about PLC it is input or output I want these level of clarification.
thank you

MartyG-RealSense, 2020-04-21:
This page looks like a good introduction to Piecewise Linear Complex (PLC):

https://wias-berlin.de/software/tetgen/plc.html

It describes PLC as an input.

thambimani, 2020-04-21:
> This page looks like a good introduction to Piecewise Linear Complex (PLC):
> 
> https://wias-berlin.de/software/tetgen/plc.html
> 
> It describes PLC as an input.

Thank you

closed 2020-05-17 by MartyG-RealSense