realsenseai/librealsense #6275: Get xyz givin pixel(v,u)

jorge1993, 2020-04-20:
Someone knows how to discover the pointcloud "pixel"(x,y,z in real world) giving a pixel (v,u) from RGB camera? 

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          |

MartyG-RealSense, 2020-04-20:
I feel that @zivsha may have done something like what you are aiming for, though in C++ instead of Python.

https://github.com/IntelRealSense/librealsense/issues/1143#issuecomment-364750000

zivsha, 2020-04-20:
[image]

MartyG-RealSense, 2020-04-21:
@zivsha   I meant that it looked as though you had done some coding regarding converting (u,v) values into coordinates.  :)

zivsha, 2020-04-21:
Been a while since I've contributed to this repo :)

jorge1993, 2020-04-22:
I resolved, with this instruction rs2_deproject_pixel_to_point

Thanks!
P.S. All of you are awesome

closed 2020-04-22 by jorge1993

MartyG-RealSense, 2020-04-22:
@jorge1993   Great news that you found a solution.  Thanks for the update!  :)