realsenseai/librealsense #6277: Problems to create a Docker image of Cubemos Skeleton Tracking SDK

mferrero82, 2020-04-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |   Ubuntu 18.04  |                                          | 
| Platform                                 | PC  |
| SDK Version                            |  cubemos-SkeletonTracking_2.3.0 | 
| Segment			| Installation | 

### Issue Description

I am trying to make a Docker image of a Cubemos Skeleton Tracking SDK installation. To do so, I start from an Ubutu 18.04 Docker image and do all the steps in the installation guide. I want to do all the steps of the installation and finally save an image of the Docker container state.

Unfortunately, I am having problems with the License activation. I got the following log:

[CUBEMOS_LOG_2][2020-Apr-20 11:24:53.715508][info]Cubemos Logging initialised
[CUBEMOS_LOG_3][2020-Apr-20 11:24:53.715675][info]Initialising the cubemos skeleton tracking v2.3.0
[CUBEMOS_LOG_4][2020-Apr-20 11:24:53.715758][debug]The variable "CUBEMOS_SKEL_SDK" points to: "/opt/cubemos/skeleton_tracking"
[CUBEMOS_LOG_5][2020-Apr-20 11:24:53.715805][info]Initialising the cubemos engine v2.2.0
[CUBEMOS_LOG_6][2020-Apr-20 11:24:53.715837][warning]Activation key not found at the provided path /root/.cubemos/skeleton_tracking/license/activation_key.json
[CUBEMOS_LOG_7][2020-Apr-20 11:24:53.715866][warning]Activation will be attempted by connecting to the cubemos activation server.
[CUBEMOS_LOG_8][2020-Apr-20 11:24:53.738068][debug]Intel Core Processor Identifier: 6820; Core Version: 7
[CUBEMOS_LOG_9][2020-Apr-20 11:24:53.842391][error]Hardware information reading failed: basic_string::_M_construct null not valid
[CUBEMOS_LOG_10][2020-Apr-20 11:24:53.842668][error]Exception thrown in file /home/sttl/Workspace/sdk/skeleton_tracking/dep/cubemos_core/sources_and_scripts/core/modules/engine/src/internal_api.cpp and line 134 with error message: "basic_string::_M_construct null not valid. Return code: 1"

Is there anything I can do to finish successfully the installation in a Docker container?
Is it possible to use Cubemos Skeleton Tracking SDK with Docker?

MartyG-RealSense, 2020-04-20:
I recall a case where a RealSense user had a similar problem with the licence when using Docker with the **Nuitrack SDK** skeleton tracking software.  They were able to use the free trial version with Docker but that was not a suitable solution.  As far as I know, there still is not a solution for using Docker with Nuitrack's licenced versions.

It would probably be best to ask Cubemos about it directly at **support@cubemos.com**

mferrero82, 2020-04-21:
Thank you very much for your fast reply. I will contact Cubemos support.

closed 2020-04-26 by dorodnic

mhudnell, 2020-07-01:
@mferrero82 were you able to get the cubemos sdk running in docker? Am thinking of trying to do the same.