realsenseai/librealsense #6280: Build librealsense 2.33.1 for Win7 or higher

svebert, 2020-04-21:
Hi!
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D415 | 
| Firmware Version                   | (5.11.6.250) | 
| Operating System & Version |   {Win10  | 
| Kernel Version (Linux Only)    |                                  | 
| Platform                                 | PC  |
| SDK Version                            |  2.33.1 }                          | 
| Language                            |  C++ }                          | 
| Segment			|  others                       | 

### Issue Description
i have trouble linking the librealsense.lib which I did build from source with vc140_xp.
CMake build configuration: Everything off, but EASYLOGGINGPP and FORCE_RSUSB_BACKEND.
My aim is it to have a static library for windows 7 and higher. Thats why i use FORCE_RSUSB_BACKEND. 
1) I hope this is correct?
2) Why do I have to specify LIBUSB_INC and LIBUSB_LIB? In my understanding FORCE_RSUSB_BACKEND leads to librealsense using WinUSB?

I found setupapi.lib, winusb.lib and cfmgr32.lib in the Windows Driver Kits. Everything works fine if i use the files from the WDK for Windows 8.1 or higher (winver6.3). (I did not test winver6.2 eg. Windows 8 and I did only tests on Windows 10.)

3) If I use cfgmgr32.lib  dedicated for Windows 7 I get linker errors. Why is that? What I am doing wrong? The cfgmgr32.lib was found in the "WDK for Windows 7" which has the name DDK (for example here: https://www.microsoft.com/en-us/download/details.aspx?id=11800). Actually that version of cfgmgr32.lib does not provide all needed functions. I get linker errors regarding DEVPROPERTY in my application. As I understood FORCE_RSUSB_BACKEND ist for Windows7, so I would think, that I should use a cfgmgr32.lib made for Windows 7. winusb.lib and setupapi.lib do not show these version dependent problems.

MartyG-RealSense, 2020-04-22:
Apologies for the delay in responding, as I was considering your case carefully.  

I note that you had a failed build when using FORCE.  Could you try the CMake instruction below please if you have not done so already:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

svebert, 2020-04-23:
Hi!
thank you for picking up my issue.
I tried to run your command but I had to change it a little bit:
**cmake ..\librealsense-2.33.1\librealsense-2.33.1 -G "Visual Studio 14 2015 Win64" -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_SHARED_LIBS=false**

I changed the generator, as I do not have "Visual Studio 15 2017" set up on my system (which is what cmake tries as default  (?)). I also added -DBUILD_SHARED_LIBS=false, as I want to build the .lib and not the .dll.
With the command above I am able to build librealsense.lib and librealsense-file.lib in the generated VS solution. I also tried the same without -DBUILD_SHARED_LIBS=false, which also compiles.
But! When I use the realsense.lib in my project and compile with v140 x64, I get the following linker error:

> Creating library D:\Projects\myDir\Release MT\x64\MyProj.lib and object D:\Projects\myDir\Release MT\x64\MyProj.exp
> 1>  realsense2.lib(win-helpers.obj) : error LNK2019: unresolved external symbol GUID_DEVINTERFACE_USB_DEVICE referenced in function "bool __cdecl librealsense::platform::get_usb_device_descriptors(unsigned long,unsigned short,unsigned short,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &,enum librealsense::platform::usb_spec &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &)" (?get_usb_device_descriptors@platform@librealsense@@YA_NKGGAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEAV34@AEAW4usb_spec@12@11@Z)
> 1>  D:\Projects\myDir\Release MT\x64\MyProj.dll : fatal error LNK1120: 1 unresolved externals
> 1>Done building target "Link" in project "MyProj.vcxproj" -- FAILED.

The next problem is, that I need to use v140_xp and not v140. If I try the following cmake command:
**cmake ..\librealsense-2.33.1\librealsense-2.33.1 -G "Visual Studio 14 2015 Win64" -T v140_xp -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_SHARED_LIBS=false**
I get linker errors when building the generated VS solution:

> 6>d:\projects\librealsense-2.33.1\librealsense-2.33.1\src\win\win-helpers.h(75): error C2061: syntax error: identifier 'DEVPROPKEY' (compiling source file D:\Projects\librealsense-2.33.1\librealsense-2.33.1\src\win\win-helpers.cpp)
> 6>D:\Projects\librealsense-2.33.1\librealsense-2.33.1\src\win\win-helpers.cpp(12): fatal error C1083: Cannot open include file: 'usbioctl.h': No such file or directory
> 6>d:\projects\librealsense-2.33.1\librealsense-2.33.1\src\winusb\endpoint-winusb.h(9): fatal error C1083: Cannot open include file: 'winusb.h': No such file or directory (compiling source file D:\Projects\librealsense-2.33.1\librealsense-2.33.1\src\winusb\interface-winusb.cpp)
> 6>D:\Projects\librealsense-2.33.1\librealsense-2.33.1\src\win/win-helpers.h(75): error C2061: syntax error: identifier 'DEVPROPKEY' (compiling source file D:\Projects\librealsense-2.33.1\librealsense-2.33.1\src\winusb\device-winusb.cpp)
> 6>d:\projects\librealsense-2.33.1\librealsense-2.33.1\src\winusb\handle-winusb.h(9): fatal error C1083: Cannot open include file: 'winusb.h': No such file or directory (compiling source file D:\Projects\librealsense-2.33.1\librealsense-2.33.1\src\winusb\device-winusb.cpp)
> 6>D:\Projects\librealsense-2.33.1\librealsense-2.33.1\src\win/win-helpers.h(75): error C2061: syntax error: identifier 'DEVPROPKEY' (compiling source file D:\Projects\librealsense-2.33.1\librealsense-2.33.1\src\winusb\enumerator-winusb.cpp)
> 6>d:\projects\librealsense-2.33.1\librealsense-2.33.1\src\winusb\request-winusb.h(11): fatal error C1083: Cannot open include file: 'winusb.h': No such file or directory (compiling source file D:\Projects\librealsense-2.33.1\librealsense-2.33.1\src\winusb\request-winusb.cpp)
> 6>D:\Projects\librealsense-2.33.1\librealsense-2.33.1\src\win/win-helpers.h(75): error C2061: syntax error: identifier 'DEVPROPKEY' (compiling source file D:\Projects\librealsense-2.33.1\librealsense-2.33.1\src\winusb\messenger-winusb.cpp)
> 6>d:\projects\librealsense-2.33.1\librealsense-2.33.1\src\winusb\handle-winusb.h(9): fatal error C1083: Cannot open include file: 'winusb.h': No such file or directory (compiling source file D:\Projects\librealsense-2.33.1\librealsense-2.33.1\src\winusb\enumerator-winusb.cpp)
> 6>d:\projects\librealsense-2.33.1\librealsense-2.33.1\src\winusb\handle-winusb.h(9): fatal error C1083: Cannot open include file: 'winusb.h': No such file or directory (compiling source file D:\Projects\librealsense-2.33.1\librealsense-2.33.1\src\winusb\messenger-winusb.cpp)
> 6>

MartyG-RealSense, 2020-04-23:
I apologise for the delay in responding.  I was carefully considering your situation.

I found an example of a CMake build statement for Visual Studio 14 2015 in a Librealsense file, so you could try this format:

**cmake .. -G "Visual Studio 14 2015 Win64"**

MojamojaK, 2020-05-01:
I have dealt with a similar problem earlier before.

If you want to use the `-DFORCE_RSUSB_BACKEND=true` option, simply building librealsense from source does not install the required driver (IntelRealSense_D400_series_win7.inf) for you, which won't allow you to enumerate through devices.

You would have to install the driver from the installer from
**Releases** > **2.33.1** (or any desired version) > **Assets** > [**Intel.RealSense.SDK-WIN7-2.33.1.1360.exe**](https://github.com/IntelRealSense/librealsense/releases/download/v2.33.1/Intel.RealSense.SDK-WIN7-2.33.1.1360.exe)
and Install the Windows 7 driver (yes you have to be admin).

During installation, you only have to tick the `Intel Realsense Win7 Drivers (blah blah blah)` option as show below.
[image]

If you ever decide to use revert back to using `-DFORCE_RSUSB_BACKEND=false`, you will have to uninstall the driver. See #6033.

MartyG-RealSense, 2020-05-17:
Hi @svebert    Do you require further assistance with this question please, or can this case be closed?   Thanks!

closed 2020-05-24 by MartyG-RealSense