realsenseai/librealsense #6282: rs-post-processing Sample - readme.md

PerceptMD, 2020-04-21:
No big thing, but there is a small difference between the code in readme against the cpp file.

` Line 37:   filter_options(const std::string name, rs2::filter& filter);`
in https://github.com/IntelRealSense/librealsense/blob/master/examples/post-processing/rs-post-processing.cpp

`    filter_options(const std::string name, rs2::process_interface& filter);`
https://github.com/IntelRealSense/librealsense/tree/master/examples/post-processing

MartyG-RealSense, 2020-04-21:
Thanks so much for your report!  I have labelled this post with the 'Documentation' tag so that it can be tracked by RealSense team members responsible for documentation.

MartyG-RealSense, 2022-05-16:
Hi @PerceptMD  As the documentation change request that you made in 2020 is now 2 years old, can you confim whether you still believe that the requested change should be made, please?  Thanks!

closed 2022-05-28 by MartyG-RealSense