realsenseai/librealsense #6283: STL4015: The std::iterator class template is deprecated in C++17

YHabotAetrex, 2020-04-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415                                                      | 
| Firmware Version                   | 05.11.15.00                                            | 
| Operating System & Version |  Win 10                                                   | 
| Platform                                 | PC                                                          |
| SDK Version                           |  2.29.0                                                   | 
| Language                               |  C++ Visual Sudio 2019 v 16.5.4           | 

### Issue Description
Getting a build error in :
\Intel RealSense SDK 2.0\include\librealsense2\hpp\rs_frame.hpp(1102,38): error C4996: 'std::iterator<std::forward_iterator_tag,rs2::frame,ptrdiff_t,rs2::frame *,rs2::frame &>': warning STL4015: The std::iterator class template (used as a base class to provide typedefs) is deprecated in C++17. (The <iterator> header is NOT deprecated.) The C++ Standard has never required user-defined iterators to derive from std::iterator. To fix this warning, stop deriving from std::iterator and start providing publicly accessible typedefs named iterator_category, value_type, difference_type, pointer, and reference. Note that value_type is required to be non-const, even for constant iterators. You can define _SILENCE_CXX17_ITERATOR_BASE_CLASS_DEPRECATION_WARNING or _SILENCE_ALL_CXX17_DEPRECATION_WARNINGS to acknowledge that you have received this warning.

MartyG-RealSense, 2020-04-21:
I carefully considered how to label your request to determine how the RealSense team track it.  As others have not reported it as something that is breaking their builds, I have labelled it as "Enhancement" (a 'feature request').  This case will remain open so that it can be tracked.  I hope that this is a satisfactory approach for you.

MartyG-RealSense, 2020-06-14:
There were no other reports from RealSense users of this problem occurring in the 2 months after this case was created.  I will therefore close the case.

danilogr, 2020-06-16:
Hey @MartyG-RealSense, would you mind re-opening this?

My info is similar to @YHabotAetrex, I am using RealSense2 on Windows with a C++17 compiler (Visual Studio v142).

I would also consider renaming this issue. `std::iterator` class is *not* deprecated in C++17. C++17 is deprecating *inheriting* from `std::iterator`.

The solution, for now, is to define `_SILENCE_ALL_CXX17_DEPRECATION_WARNINGS`, but it would be great to have C++17 compatible code. This blog has a really nice guide on the matter: https://www.fluentcpp.com/2018/05/08/std-iterator-deprecated/

MartyG-RealSense, 2020-06-16:
@danilogr   Ok, I will re-open it and take another look at it tomorrow, as the details that you have kindly provided will be helpful for me to possibly file a bug report, given that it is a replicable problem (experienced by more than one RealSense user) and generates warnings.

ev-mp, 2020-06-17:
@danilogr , @MartyG-RealSense , I will re-classify this as enhancement request - the SDK is explicitly targeted to C++11 standard. 
Being compatible with the next versions of the language is envisaged, yet it is not the same as not being compliant with specifications.

MartyG-RealSense, 2020-06-17:
@ev-mp  Thanks very much.   This was a case where classifying it as an enhancement or bug was not so clear.  Thanks for the clarification!

MartyG-RealSense, 2021-10-20:
Hi @YHabotAetrex and @danilogr   The issue in this case has now been addressed with a Pull Request that has been merged into the RealSense SDK.  I will therefore now close this case.  Thanks very much for your reports!