realsenseai/librealsense #6285: [Windows] Old Microsoft drivers used for D415 camera

BryanEmmerson, 2020-04-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  N/A                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.34.0                          | 
| Language                            |  N/A                          | 
| Segment			|  Robot                          | 

### Issue Description

Windows is using a Microsoft driver from 2006 for the RealSense camera on a number of my machines, see Device Manager window below:

[image]

I have installed the RealSense SDK on at least three machines and all of them have used that Microsoft driver by default.

I noticed that when he was helping with issue 6254, @MartyG-RealSense said the below drivers were correct:

[image]

_Originally posted by @alexander-jh in https://github.com/IntelRealSense/librealsense/issues/6254#issuecomment-614866100_

- How come the Intel driver is not installed with the SDK?
- Are there performance/behaviour differences between the drivers?

If any one else is interested, the driver download is https://downloadcenter.intel.com/download/28723/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10?v=t.

Thanks!

MartyG-RealSense, 2020-04-22:
@BryanEmmerson   Hi there!  The driver that you linked to is not one of the main camera drivers.  It is one that is used to make the camera compatible with hardware supporting Microsoft Universal Windows Platform (UWP).  An example of such hardware is the Microsoft Hololens 'mixed reality' headset.

It is common practice for drivers to be outside of software that uses them.  It makes it easier to update the drivers over the internet or with a driver installer program without having to download and install the entire software package again.

It is important to make a distinction between the RGB / depth camera drivers (which are stored on the computer) and the **firmware** camera driver, which is stored inside the camera.  Firmware is commonly inside the hardware to regulate its essential functions.  This is why a device can stop working (or become "bricked") if a bad firmware is released by a manufacturer.

Intel makes changes to drivers through firmware updates that are stored in the camera instead of changing the RGB and depth drivers.  An advantage of this is that the update gets carried inside the.camera to whatever device the camera is attached to.

Since the launch in 2019 of Intel's cross platform firmware updated system, firmware has also been bundled inside RealSense SDK releases, so that the camera can be updated to the latest recommended firmware for that particular SDK version.  This provides a way to update the RealSense SR300 / SR305 and the T265, as at the time of writing this, only the 400 Series cameras have a separate independent firmware download page and installer tool (**rs-fw-update**).

Finally, it is my experience that it is common to see very old dates on some Windows drivers.  It does not affect their operation.

BryanEmmerson, 2020-04-22:
Thanks for that! The part where I am confused is the distinction between the device drivers on the operating system versus whatever the SDK is doing when I am interacting with it.

I assume that other than making the device conform to some UWP interface, there no performance penalty for using the Microsoft vs. Intel driver.

MartyG-RealSense, 2020-04-22:
The UWP driver is not required to be installed to use the camera.  It is just for using the camera with hardware such as Hololens that has a UWP interface.  In most cases on Windows the camera should be "plug and play" - insert it into a USB port and the RGB and Depth drivers are automatically found and installed on that computer.

The cameras come with an older firmware already installed so that the camera can work straight away.  You can then manually update the firmware stored inside the camera hardware to a recent version using the RealSense Viewer program (or the **rs-fw-update tool if you have a 400 Series camera).

BryanEmmerson, 2020-04-22:
> The UWP driver is not required to be installed to use the camera. It is just for using the camera with hardware such as Hololens that has a UWP interface. In most cases on Windows the camera should be "plug and play" - insert it into a USB port and the RGB and Depth drivers are automatically found and installed on that computer.
> 
> The cameras come with an older firmware already installed so that the camera can work straight away. You can then update the firmware to a recent version using the RealSense Viewer program (and also update using the **rs-fw-update** tool if you have a 400 Series camera).

The camera firmware wasn't my focus for this issue as I've already been testing various versions for a stability problem I'm having. 

I asked the question as I found your comment on the Intel UWP drivers and thought that the device drivers could be to blame. 

Thanks for your help!

closed 2020-04-22 by BryanEmmerson