realsenseai/librealsense #6286: 2.34.0 : T265 `pipe.start()` stuck

pjessesco, 2020-04-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 2.34.0 | 
| Operating System & Version |   Windows 10  |                                    | 
| Platform                                 | PC  |
| SDK Version                            |   2.34.0                           | 
| Language                            | C++                           | 
| Segment			|  AR                          | 

### Issue Description

Our project uses 2 cameras, D435I and T265, and it worked well on 2.31. Because we want to improve our project's accuracy, we upgraded SDK version to 2.34, refer to its release note, that improves IMU. However it doesn't work after upgrade SDK.

Below is the reduced code that reproduces the issue.

	// License: Apache 2.0. See LICENSE file in root directory.
	// Copyright(c) 2015-2017 Intel Corporation. All Rights Reserved.

	#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
	#include "example.hpp"          // Include short list of convenience functions for rendering

	int main(int argc, char * argv[]) try
	{
		rs2::context                ctx;            // Create librealsense context for managing devices
		std::vector<rs2::pipeline>  pipelines;

		std::cout << "Starting device" << std::endl;
		// Start a streaming pipe per each connected device

		for (auto&& dev : ctx.query_devices())
		{
			std::cout << dev.get_info(RS2_CAMERA_INFO_NAME) << std::endl;
			std::cout << dev.get_info(RS2_CAMERA_INFO_FIRMWARE_VERSION) << std::endl;
			rs2::pipeline pipe(ctx);
			rs2::config cfg;
			std::cout << "enable device" << std::endl;
			cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
			std::cout << "start pipe" << std::endl;
			pipe.start(cfg);
			pipelines.emplace_back(pipe);
			std::cout << "Done " << dev.get_info(RS2_CAMERA_INFO_NAME) << std::endl;
			std::cout << "------------------------------" << std::endl;
		}

		std::cout << "Device started" << std::endl;

		return EXIT_SUCCESS;
	}
	catch (const rs2::error & e)
	{
		std::cout << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
		return EXIT_FAILURE;
	}
	catch (const std::exception & e)
	{
		std::cout << e.what() << std::endl;
		return EXIT_FAILURE;
	}

and here is the output.

    Starting device
    Intel RealSense D435I
    05.12.03.00
    enable device
    start pipe
    Done Intel RealSense D435I
    ------------------------------
    Intel RealSense T265
    0.2.0.926
    enable device
    start pipe

Process stucks after print `start pipe`, and terminated.

piotr-tanski, 2020-04-22:
I confirm the same issue on 2.33.1.

Generally, pipeline::start(config) hangs if specific device is enabled in rs2::config:
`rs2::config config;
config.enable_device(deviceSerialNumber);
config.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
const auto profile = m_pipeline.start(config);`

While this works just fine:
`
rs2::config config;
config.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
const auto profile = m_pipeline.start(config);`

Worth to notice that in both cases
`profile.get_device().get_info(RS2_CAMERA_INFO_SERIAL_NUMBER) == deviceSerialNumber`

pjessesco, 2020-04-23:
Hi @piotr-tanski, thanks for your advice and code snippet. I fixed my code :

    for (auto&& dev : ctx.query_devices())
	{
		rs2::pipeline pipe(ctx);
		rs2::config cfg;
		cfg.enable_all_streams();
		pipe.start(cfg);
		pipelines.emplace_back(pipe);
	}

and it works fine.

However, the issue still remains, since the code that doesn't work is from the [example](https://github.com/IntelRealSense/librealsense/blob/bc64e52c9e128aa82cc9f25a14283bbe2bcf8828/examples/tracking-and-depth/rs-tracking-and-depth.cpp#L38-L45).

Alnilam1305, 2020-04-24:
I struggle with the same problem on T265 only with 2.33.1. When I communicate with the device, e.g. to read or set sensor options as it is done in the rs-sensor-control example and then try to start the pipe for getting sensor data, pipe.start always blocks. When I jump over communication with the device the same code runs. 
This is very annoying. Is there a workaround for this? Obviously it seems to run in the realsense viewer somehow.

BryanEmmerson, 2020-04-28:
Hi @pjessesco,

When you say it freezes, what is the behaviour? Does it cause the camera to stop working until you do a power cycle?

I ask as I'm having a similar issue with a D415 that throws errors on a pipeline start.

VladMVLX, 2020-04-29:
Having the same issue while connecting T265 and D435i together time if I do:
`rs2::pipeline pipe(_context);
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);       
pipe.start(cfg);`

then it hangs at pipe.start, but if I do:
`rs2::pipeline pipe(_context);
rs2::config cfg;
cfg.enable_all_streams();       
pipe.start(cfg);`
it hangs at ` pipe.wait_for_frames`

realsenseSDK 2.24.0 , sample rs-tracking-and-depth is not working as well having the same issue

Alnilam1305, 2020-04-29:
I was somehow successful by using dev.hardware_reset(); after configuring the device options and before starting the pipe. Doing so the pipe works, but i am not quite sure, if the settings are still all there. In a first quick check it seemed for me that they are still there.

kamran00f, 2020-05-04:
I have the same issue with SDK 2.34.0. I run the following code:

`
            std::vector<rs2::pipeline *> pipelines;
            rs2::pipeline *pipe = new rs2::pipeline(ctx);
            rs2::config cfg;
            std::string sn1("948422110659"); // also tried 0000948422110659
            cfg.enable_device(sn1);
            cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
            std::cout << "starting pipeline" << std::endl;
            pipe->start(cfg);
            std::cout << "pipeline started" << std::endl;
            pipelines.emplace_back(pipe);
`

 pipe->start(cfg) gets stuck and after about 15 seconds it throws an exception with the following message:
`
RealSense error calling rs2_pipeline_start_with_config(pipe:0x7edd28, config:0x814908):
    No device connected
`

I noticed if I do not call cfg.enable_device(sn1); it will work fine, but I want to get data from two T265 cameras at the same time, therefore I need two pipelines with one camera enabled in each.
I tried this example:
[https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam] (https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam)
I had the same error. Is there any known way or preferably example code to use two 265 cameras simultaneously?

VladMVLX, 2020-05-04:
> I have the same issue with SDK 2.34.0. I run the following code:
>
> ```
>  std::vector<rs2::pipeline *> pipelines;
>  rs2::pipeline *pipe = new rs2::pipeline(ctx);
>  rs2::config cfg;
>  std::string [...]

For me, as temporary solution, it helped to roll back to 2.29.0 version everything works just fine using it.

drapado, 2020-05-04:
The error persist in 2.34.1

kamran00f, 2020-05-04:
> For me, as temporary solution, it helped to roll back to 2.29.0 version everything works just fine using it.

Thanks, rolling back to SDK 2.29.0 worked for me as well.

pjessesco, 2020-06-19:
FYI : multicam example is [fixed](https://github.com/IntelRealSense/librealsense/blob/475a62d08dcc5a7959353b0b6954018597d6bc83/examples/multicam/rs-multicam.cpp#L26), so closing this issue.

closed 2020-06-19 by pjessesco