realsenseai/librealsense #6287: Enabling hand tracking and point cloud streaming on Unity simultaneously

egemenertugrul, 2020-04-22:
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC / Unity 2019.3.1f 64-bit |
| SDK Version                            |  2016 R2                           | 
| Language                            |  C# / Unity                         | 
| Segment			|  VR/AR/Others                          | 

### Issue Description
I'm currently trying to enable both hand tracking and live point cloud streaming on Unity. In the RSSDK folder, I've managed to find _../framework/Unity/**FF_HandsAnimation**_ and _../wrappers/unity/Assets/RealSenseSDK2.0/Scenes/Samples/**PointCloudDepthAndColor.unity**_ . Obviously, trying to run both scenes in a single scene throws an error about the Realsense camera being already in use. I tried to look into the code of Point Cloud example, however, I could not find a way to enable hand tracking from there. It seems as if that particular demo is dedicated to streaming point cloud only which does not let me enable other features, such as hand tracking in my case.

I've also tried importing _../framework/Unity/**UnityToolkit.unitypackage**_ and running _**Sample6 - Point Cloud.unity**_ but this led Unity to freeze every time, leaving me no choice but end process from the task manager (other demos work fine).

I would really appreciate if someone could at least show me a way to make it work. Thank you.

MartyG-RealSense, 2020-04-22:
The 2016 RealSense SDK had a totally different architecture to Librealsense, unfortunately.  At that time, the first version of Librealsense (now known as Legacy Librealsense) was developed and used alongside the Windows SDK but the two SDKs had no compatibility at all.

I used to be a specialist in Unity development in the old 2016 Windows SDK though, so hopefully I can provide some useful guidance.

If you were trying to merge **FF_HandsAnimation** for the 2016 SDK with **PointCloudDepthAndColor** for the current RealSense SDK 2.0 (Librealsense2) then they could not work together in the same program.  They access completely different DLL library files.  

And even if code for the 2016 and current SDKs was in the same script and you had both sets of DLLs installed, it probably still would not work as both the hand tracking (2016) and point cloud (current) code would be trying to open incompatible pipelines to access the same SR300 camera at the same time.  It would be like two video chat programs trying to both stream video from the same camera at the same time.

I recall that during the 2016 years, I used Unity 4.6 for RealSense Unity development.  You may find that you can get the point cloud sample to work in that old Unity version, though obviously it is not an ideal solution if you wanted to use the camera in a modern Unity project.

https://unity3d.com/get-unity/download/archive?_ga=2.19143099.269430220.1587578488-80167007.1577007397

I also used Unity 5 with the 2016 SDK after Unity 5 launched, though I cannot recall if the point cloud sample worked with it.

egemenertugrul, 2020-04-23:
@MartyG-RealSense Thank you for your reply. I will test if it works with the previous versions of Unity and let you know. As you mentioned, my initial thought was to make it work with the recent versions (> 2019) to support new XR features of Unity.

In addition, I tried to figure out what freezes Unity in _**Sample6 - Point Cloud.unity**_ demo, but Visual Studio 2019 debugger would lose track of the Unity process as soon as I hit the play button. It is possible to see point cloud appearing on the scene occasionally (one frame at a time), which led me to think that the incoming Realsense stream bottlenecks the main Unity thread. I should do further investigation in order to find the cause of the problem with the recent Unity versions and devise a workaround.

MartyG-RealSense, 2020-04-23:
In the old days, a script editor called MonoDevelop was used to edit and compile scripts.  It got replaced by Visual Studio from the 2018 Unity editions onward.  You used to be able to switch back to MonoDevelop but I don't think they support that in the 2019 editions any more.

MartyG-RealSense, 2020-05-17:
Hi @egemenertugrul   Do you require further assistance wit this question please, or can the case be closed?  Thanks!

closed 2020-05-24 by MartyG-RealSense