realsenseai/librealsense #6292: RealSense L515 documentation

talbenach, 2020-04-23:
I am trying to find information on the L515 camera beyond the datasheet Intel provides. Some questions that we have are:
1. The D4xx cameras provide the depth and RGB streams separately, and require the Align function to match IR depth to RGB XY. Is that the same with the L515 or does it provide the depth stream already aligned with the RGB image?
2. What is the camera's depth accuracy at short ranges - between 25 and 75 cm?
3. Any papers written describing "best known methods" using this camera? Specifically, in applications that require identification of tags (e.g. Aruco tags) and tracking tools movement?

MartyG-RealSense, 2020-04-23:
1.  Like the 400 Series cameras, the L515 uses the **OmniVision OV2740** imager component for RGB.  The L515 uses an IR laser beam though (the 400 Series used an OmniVision imager for IR).  The L515's MEMS component is used to scan the IR laser beam over the entire field-of-view (FOV). 

In regard to how depth is generated, the data sheet states:

"The L515 vision ASIC will process the data from the reflected beam captured by the photodiode and will output a depth point representing the accurate distance of a specific point in the scene
from the camera.  Aggregation of the depth points will generate a point cloud depth data representing the full scene".

If the L515 is like the 400 Series in terms of how RGB and depth alignment is handled: if the RGB sensor is separate from the IR laser component and they are in different physical locations on the camera, I would expect depth to color alignment to still need to be configured rather than aligned by default.  @dorodnic could clarify how the L515 handles depth-color alignment.

2.  In addition to the L515 data sheet document, there is also an official information web-page.

https://www.intelrealsense.com/lidar-camera-l515/

The FAQ at the base of the page is not working at the time of writing this.  When it was working though, it stated that the L515 has a depth error average of **< 5 mm at 1 m range**.

3.  There are not papers currently available for the L515.  Though a company called **LUMOplay** recently did an analysis on YouTube of the L515 with a pre-release sample of the camera though, and this may provide useful insights ahead of the L515's release.

https://github.com/IntelRealSense/librealsense/issues/6214

dorodnic, 2020-04-23:
Hi @talbenach 
L515 also has a separate RGB sensor
Regarding accuracy I think the data-sheet is the only source of official information. 
With the release of the product we will continue to publish white-papers like we do for the D400.

JiamingSuen, 2020-04-26:
I also have some questions regarding the L515 LiDAR camera.
1. What is the estimated shipping date for pre-orders made before March 31 given the current COVID-19 status?
2. Is it possible to provide some sample rosbags of this sensor as addressed in https://github.com/IntelRealSense/librealsense/issues/5430#issuecomment-589986552?

MartyG-RealSense, 2020-04-26:
1.  As far as I know, RealSense Store product shipments are not being affected by COVD-19.  When you receive your camera may depend on how early you pre-ordered.  The first pre-orders were scheduled to ship on the week of April 27.  Some weeks after pre-orders first opened, one customer said that they were quoted a ship date of June.  

Currently, pre-orders made after March 31 are scheduled to ship in the week of July 26.  Pre-orders are processed in order of when they were received.

2.  Supporting materials such as sample bag files, CAD models and white papers tend to "arrive when they arrive".  We make sure to promote the availability of new resources once they are available, but there are usually not firm time estimates available.

MartyG-RealSense, 2020-05-17:
Hi @JiamingSuen    Do you have further questions about the L515, or is it okay to close this case now, please?  Thanks!

JiamingSuen, 2020-05-17:
> Hi @JiamingSuen Do you have further questions about the L515, or is it okay to close this case now, please? Thanks!

I don't have further questions for now, but I'm not the owner of this issue though.

MartyG-RealSense, 2020-05-17:
@JiamingSuen   Thanks so much!

@talbenach  Thanks for your L515 questions!  As it has been a few weeks now without follow-up questions, I will close the case.  Feel free to create a new case if you have new questions about the L515 in future.

closed 2020-05-17 by MartyG-RealSense

talbenach, 2020-05-18:
Hi Marty,

I have a new question on the 515 camera (which probably should be a new issue anyway…).

We purchased the Intel calibration target for the D400 cameras. Will it be relevant for the L515 camera as well? Is there another method of OEM calibration of the L515 camera?

Thank you very much,

Noam

Closed #6292<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FIntelRealSense%2Flibrealsense%2Fissues%2F6292&data=01%7C01%7Cnoaml%40brainsgate.com%7C6306bf8dfce543cc423608d7fa805392%7Cba1b9a8211954a63b7107dab2a24ddaf%7C0&sdata=Up5PSYNnjQgTTTS6uo%2FmO%2BUVt5KGoq1CgWtkRhk9ab8%3D&reserved=0>.

MartyG-RealSense, 2020-05-18:
The L515 camera is designed so that it should never need recalibrating during its lifetime.

talbenach, 2020-05-18:
Thank you Marty,

Does you answer applies to the intrinsic parameters of the RGB camera as well?

MartyG-RealSense, 2020-05-18:
The information about calibration refers to the L515 retaining its depth accuracy throughout its lifespan.   The L515 uses the OV2740 imager component for the RGB like the 400 Series cameras do, but I have no information on whether the RGB imager in the L515 ever needs calibration.  

@dorodnic  Could you provide advice on this point please?

5trobl, 2021-07-30:
@MartyG-RealSense 
Does the CustomRW application for the D400 series work for the L515 as well?

5trobl, 2021-07-30:
Apparently it doesn't, alas: https://github.com/IntelRealSense/librealsense/issues/7640#issuecomment-715595242