realsenseai/librealsense #6293: D435i base on TX2 ubuntu 16.04, rs-enumerate-devices : No device detected. Is it plugged in?

lightsalt2011, 2020-04-23:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (5.11) | 
| Operating System & Version |   ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4.38-tegra                                         | 
| Platform                                 |  Jetson TX2  L4T 28.2.1 (JetPack 3.2.1) |
| SDK Version                            | v2.13.0                        | 
| Language                            | c                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I have follow https://blog.csdn.net/u013595260/article/details/86522385 https://github.com/jetsonhacks/buildLibrealsense2TX step by step, 
buildLibrealsense2TX, git check v0.9

nvidia@tegra-ubuntu:~$ lsusb -d 8086: -v | grep 'bcdUSB'
Couldn't open device, some information will be missing
  bcdUSB               3.20

[  562.252207] usb 1-3: new high-speed USB device number 3 using xhci-tegra
[  562.385482] usb 1-3: feature bit otg_vbus_off set
[  562.390273] usb 1-3: New USB device found, idVendor=8086, idProduct=0b3a
[  562.397207] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  562.404482] usb 1-3: Product: Intel(R) RealSense(TM) Depth Camera 435i
[  562.411177] usb 1-3: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[  562.419482] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work mailbox command 6
[  562.427805] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  562.439679] uvcvideo: Unable to create debugfs 1-3 directory.
[  562.446422] input: Intel(R) RealSense(TM) Depth Ca as /devices/3530000.xhci/usb1/1-3/1-3:1.0/input/input7
[  562.459265] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  562.470246] uvcvideo: Unable to create debugfs 1-3 directory.

**but  nvidia@tegra-ubuntu:~$ rs-enumerate-devices 
No device detected. Is it plugged in?**

lightsalt2011, 2020-04-23:
is there something wrong with my operation?

ev-mp, 2020-04-23:
@lightsalt2011 hello, the D435i support was [introduced with v2.16.2](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.2) of the SDK. 

Is there any particular reason that you use v2.13?

lightsalt2011, 2020-04-24:
thanks for ur reply, I have follow https://github.com/jetsonhacks/buildLibrealsense2TX  build kernel image, now version as below:
LibRealSense v2.17.1
Realsense ROS v2.1.4

both color image and depth image can be show in realsense_viewer and ros topic, but when I change the rgb(color) resolution as 1280x720 30fps, the error come up, but the 640x480 30 fps is ok

[image]

how can I fix **Incomplete video frame detected!, Size 65784 out of 1843455 bytes**  **error**

lightsalt2011, 2020-04-24:
the error log fire always without stopping, cause the fps drop maybe

dorodnic, 2020-04-26:
It's just reporting frame-drops. As far as I know, this is platform limitation, but we may make the camera perform better with more Jetson specific optimizations. 

@lightsalt2011 - please note that [jetsonhacks/buildJetsonTX2Kernel](https://github.com/jetsonhacks/buildJetsonTX2Kernel) seem to be outdated, and you can use [jetsonhacks/installRealSenseSDK](https://github.com/jetsonhacks/installRealSenseSDK). Furthermore, [installLibrealsense.sh](https://github.com/jetsonhacks/installRealSenseSDK/blob/master/installLibrealsense.sh) would require significantly less effort (compared to [buildLibrealsense.sh](https://github.com/jetsonhacks/installRealSenseSDK/blob/master/buildLibrealsense.sh))

lightsalt2011, 2020-04-27:
thanks for ur reply, I will try it on lastest jetpack version

closed 2020-04-27 by lightsalt2011