realsenseai/librealsense #6295: Starting pipeline with both IR cameras fails on Android

kudansam, 2020-04-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |  MacOS  | 
| Kernel Version (Linux Only)    | -  | 
| Platform                                 | Android 10 |
| SDK Version                            |  2.34.0 | 
| Language                            |  C++                          | 
| Segment			|  VR                          | 

I'm attempting to start a pipeline that gets data from both IR cameras simultaneously.

I generate a config and start the pipeline as follows:

```
rs2::config config;
config.enable_stream(RS2_STREAM_INFRARED, 1, 640, 480, RS2_FORMAT_Y8, 30);
config.enable_stream(RS2_STREAM_INFRARED, 2, 640, 480, RS2_FORMAT_Y8, 30);

rs2::pipeline pipe;

pipe.start(cfg);
```

This works fine on Ubuntu 18.04 but generates the following error on Android:

```
E/libc++abi: terminating with uncaught exception of type rs2::error: Couldn't resolve requests
```

Is this a bug or not supported on Android?

MartyG-RealSense, 2020-04-23:
The message "Couldn't resolve requests" may appear if the SDK is being asked to do something that is not supported or there is an error in the formatting of the statement that means the SDK cannot understand what it is being asked to do.

To access the left and right IR, I believe that you should be using the stream indexes 0 and 1 instead of 1 and 2.  It may be getting confused about being asked to access IR stream '2'.

Here is an example C++ script that accesses both left and right streams:

**rs2::config cfg;
cfg.enable_stream(RS2_STREAM_INFRARED, 0);
cfg.enable_stream(RS2_STREAM_INFRARED, 1);
rs2::pipeline pipe;
pipe.start(cfg);**

kudansam, 2020-04-23:
I've attempted using the code snipped you posted but it makes no difference to the outputted error.

MartyG-RealSense, 2020-04-23:
Is the USB port on your Android device USB 3 or USB 3, please?

kudansam, 2020-04-23:
It's a USB 3.1 port according to the specification.

MartyG-RealSense, 2020-04-23:
On a USB 2 connection with D435i, only the left IR is available.  I just tested my own D435i with the RealSense Viewer on a USB2 connection and confirmed to myself that I wasn't mis-remembering this.

[image]

It makes me wonder if your Android tablet is mis-identifying the D435i in its USB 3 port as a USB 2 device and so making right IR inaccessible to your program and triggering the "couldn't resolve request" message (as support for it is absent in USB2 mode).

kudansam, 2020-04-23:
I accessed a shell on the Android device and I think this indicates that it's connecting over USB 3:
```
DEVICE:/ # cat /sys/bus/usb/devices/3-1/manufacturer                                                                                                                                                   
Intel(R) RealSense(TM) Depth Camera 435i
DEVICE:/ # cat /sys/bus/usb/devices/3-1/version                                                                                                                                                        
 3.20
```
I'm not entirely sure of the best way to make sure of this on Android though.

MartyG-RealSense, 2020-04-23:
There was a RealSense Android user who built the **Capture** app on their Android device to check the camera speed.

https://github.com/IntelRealSense/librealsense/issues/4091#issuecomment-502467123

I am off-shift now but will be back in 7 hours from the time of writing this and be happy to continue the support conversation if you require further assistance.  Good luck!

MartyG-RealSense, 2020-05-17:
Hi @kudansam   Do you require further assistance, or is it okay to close this case now?   Thanks!

kudansam, 2020-05-18:
I couldn't get both cameras to work on Android, either running live from the camera or from a recorded rosbag file. Unfortunately I didn't have time to debug the library so had to move to another solution.

MartyG-RealSense, 2020-05-18:
Ok, thank you very much for the update.  I will close this case then.  I'm sorry that you did not achieve the results that you were aiming for.

closed 2020-05-18 by MartyG-RealSense