realsenseai/librealsense #6296: pyrealsense2 Python3.8 support

scottrpaterson, 2020-04-23:
This is related to #5777 

Pyrealsense should support Python3.8+ without needing to build manually.

MartyG-RealSense, 2020-04-23:
Thank you very much for your request.  I will label it as "Enhancement" (a Feature Request) so that it can be tracked by the RealSense team.

Wallbraker, 2020-05-18:
Any movement on this? Ubuntu 20.04 **LTS** shipped with Python 3.8 and currently it's not possible to install `pyrealsense2` on it, building from source fails as well.

MartyG-RealSense, 2020-05-18:
@Wallbraker   I have no update on this to provide, unfortunately.  I am on the Support team of the RealSense GitHub, and Feature Requests are tracked by the Developer team.

progekta, 2020-05-19:
I confirm the urgent need for python 3.8 wheel file (cp38). Building from source fails, making Ubuntu 20 unsuitable for pyrealsense...

MartyG-RealSense, 2020-05-19:
@progekta   I communicated with Intel internally yesterday about Pyrealsense2 installation issues.  I will add your message onto that communication.

MartyG-RealSense, 2020-06-14:
The feature request for Python 3.8 support for pyrealsense2 pip installation has been officially logged with Intel via a similar case.

https://github.com/IntelRealSense/librealsense/issues/6126

I will therefore close this case, as the feature request can be processed via the case linked to above.  Thanks everyone for your input!

closed 2020-06-14 by MartyG-RealSense

LivLee97, 2020-06-15:
@MartyG-RealSense Sorry, I still cannot find the officially pyrealsense2 pip installation for python 3.8

MartyG-RealSense, 2020-06-15:
@LivLee97   The file does not exist yet.  I have officially logged the 'feature request' with the developers for a pyrealsense2 wheel file for Python 3.8.   This is only a request though, and there is no timetable for when it might become available if the request is approved.   In the meantime, Python 3.8 users should compile pyrealsense2 from source code.

LivLee97, 2020-06-15:
@MartyG-RealSense I tried to build from souce code. However, it does not fit Ubuntu 20.04. I want to know one solution to solve this problem.

MartyG-RealSense, 2020-06-15:
Officially, Librealsense is only currently supported up to Ubuntu 18.04 LTS.  It is very usable on 20.04 if its source code is built with the **RSUSB** CMake installation method.

I wonder if you could build the Python bindings as part of the RSUSB build statement by including the **-DBUILD_PYTHON_BINDINGS=true** build flag.  For example, using this build statement in the Build folder of Librealsense:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

AndreV84, 2020-08-11:
pip install pyrealsense2 installs something that can not be loaded with python3, but python2; any advise? to build from source?

MartyG-RealSense, 2020-08-12:
@AndreV84   pip install pyrealsense2 will not work with Arm based devices such as Raspberry Pi and Nvidia Jetson.  

A Jetson Xavier user of pyrealsense2 shared his solution for adjusting CMake so that it would work with Python 3.

https://github.com/IntelRealSense/librealsense/issues/6980#issuecomment-666858977

AndreV84, 2020-08-12:
@MartyG-RealSense 
Thank you for folowing up!
I found that on x84_64 python3 pyrealsense can be installed with
`python3 -m pip install pyrealsense2`
it creates a file with a long name like cpython---36---pyrealsense.so or similar; that can be copied to anywhere in a way so that any python3 import will work from there

MartyG-RealSense, 2020-08-12:
@AndreV84   Thanks very much for your own solution, I have made a note of it    :)

siddharthcb, 2020-09-15:
hi @MartyG-RealSense 
is the library available for python3.8 now?

MartyG-RealSense, 2020-09-15:
Hi @siddharthcb   The pyrealsense2 wrapper has always been usable on Python 3.8, but you have to build it from source code because there is not yet a pip package for 3.8 (only up to Python 3.7 at the time of writing this).

siddharthcb, 2020-09-15:
any clue when the pip package will be available for 3.8?

MartyG-RealSense, 2020-09-15:
There is currently no information available about a timeline for a pip install package for 3.8.  An official feature request was filed this year for the creation of one though.

karlita101, 2020-09-22:
Can someone please clarify ( complete newbie here). I am currently running Visual Studio **CODE** ... is doing this whole wrapper thing for python 3.8 only possible with VS only?

MartyG-RealSense, 2020-09-22:
Hi @karlita101   If you have a PC (that likely has a type of processor chip called x86 or x64) then you can install the Python wrapper using a method called **pip install**.

The Python wrapper can also be built from source code with the **CMake** tool..  

These two methods are described in the installation section of the documentation for the Python wrapper.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#installation

maxstrobel, 2020-12-16:
@MartyG-RealSense: Any news or timeline available now?

MartyG-RealSense, 2020-12-16:
Hi @maxstrobel   No sorry, no news to provide.

tomgoddard, 2021-02-09:
Python 3.9 and 3.8 have been released and yet pyrealsense2 on PyPi only offers Python 3.7.  Seems that Intel no longer supports RealSense for Python.   This ticket which is closed seems to have the most recent info, but there is another ticket #6126 that is open.

MartyG-RealSense, 2021-02-09:
Hi @tomgoddard   Python support for RealSense has not been abandoned.  

In the link below, an explanation is offered for the time taken to implement support for Python versions newer than 3.7.

https://github.com/IntelRealSense/librealsense/issues/8229#issuecomment-767348365

A method for hex-editing the Python 3.7 pip **whl** file was also suggested in that discussion by a RealSense community member.

https://github.com/IntelRealSense/librealsense/issues/8229#issuecomment-767115385

maxstrobel, 2021-02-09:
@MartyG-RealSense 
Thanks for the response. Nevertheless, I think for most users it would be quite important to get access to packages for current Python releases via pip.

This would bring us a lot of benefits:
- less error prone
- simpler to use
- simpler to integrate into existing software frameworks
- less overall effort (packages built only once centrally, not by every user)

MartyG-RealSense, 2021-02-09:
Thanks @maxstrobel   I hope that the explanation in the provided link demonstrates that Intel remains commited to RealSense support in Python.

tomgoddard, 2021-02-09:
Hi @MartyG-RealSense.  You have a strange notion of "support" when Intel is only providing pyrealsense2 for a 2019 version of Python (3.7) two behind the current Python version that almost no Python developers use, and provide complex build scripts using PyBind11 that have only been tested with Python 3.7.  I am sure I could build pyrealsense2 as I write Python and C++ code for a living, but it does not make sense to work on an abandoned product, I can't afford to take over maintenance of Intel's code.

Maybe MartyG and Intel don't realize but they have abandoned Python RealSense support.  You can be sure there are almost no developers using RealSense with Python when they impose such large burdens and costly burdens on the developer.

The proposal to hack the pyrealsense2 binary is just laughable.  Python 3.9 and 3.7 are not ABI compatible -- you can trick Python 3.9 to accept the 3.7 pyrealsense2 by hacking the binary and then send tens of thousands of dollars down the drain as you try to debug the unpredictable crashes your customers experience.

MartyG-RealSense, 2021-02-09:
The hex editing method was provided by a RealSense community member to assist a RealSense user who required at least a temporary workaround for Python 3.9 due to project deadlines.  It was not an official Intel-provided solution. 

In that particular case, building the pyrealsense2 wrapper from source (which can be used to install pyrealsense2 for Python versions newer than 3.7) was not compatible with the RealSense user's project needs as they required a pip-based solution.  Python continues to be strongly used by RealSense developers.

maxstrobel, 2021-02-09:
I would really need some kind of timeline, if and when we can expect support for newer Python releases.

If it is simple to generate bindings for newer versions as in #8229, it can also be easily packaged and distributed via pypi. If not, we have a fundamental problem.

MartyG-RealSense, 2021-02-09:
Other than the update to pybind11 that has been merged into the SDK in pull request https://github.com/IntelRealSense/librealsense/pull/8176 I have no further information currently available about version updates to the pyrealsense2 wrapper's pip packages.

tomgoddard, 2021-02-09:
@maxstrobel
Python 3.8 was released in October 2019, and Python 3.9 was released in October 2020 while pyrealsense2 is only built for Python 3.7 (released June 2018) on PyPi.

In #6126 and #6296  a Python 3.8 PyPi package is requested and @MartyG-RealSense made an enhancement request in April 2020.  The  story after that is always the same "Nothing new to report", and "RealSense strongly supports Python".  It's a joke.  The strong support means you can try to compile it yourself although you are likely to run into problems since the RealSense team have apparently not tried to compile it for Python 3.8 or 3.9.  So you can try to update the RealSense build if you have the time.

Sadly this is a common fate of boutique hardware.  Intel puts lots of effort to making a nice piece of hardware and then does not provide usable software to control the device.  Then they are surprised when the product fails because few people developed for it.  It would be useful if @MartyG-RealSense would relay the problem to someone who can salvage this but it sounds like he does not have the power to do that.

MartyG-RealSense, 2021-03-21:
Hi everyone,

I wanted to update you on progress.  SDK version **2.43.0** has just been released and has the following comment in its release notes:

**[Python] Generate wheels for Python 3.8 + 3.9   
Package classifiers updated to support python 3.8 + 3.9** 

[image]

Vicroni, 2022-11-02:
Do you believe you could provide a classifier for python 3.10?

AndreV84, 2022-12-09:
@MartyG-RealSense 
any chance we could use realsense on wsl ubuntu?
[image]

MartyG-RealSense, 2022-12-09:
@AndreV84  A RealSense user at https://github.com/IntelRealSense/librealsense/issues/10506#issuecomment-1125860744 had camera detection problems when using WSL2.  They managed to detect the camera with Windows 11 after adjusting some settings, but did not succeed with Windows 10.