realsenseai/librealsense #6301: D435 stops working and SDK hangs

EinarBjorn, 2020-04-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   Win10 | 
| Kernel Version (Linux Only)    |                                    | 
| Platform                                 | PC |
| SDK Version                            |  FRAMOS-librealsense2 2.29.8 | 
| Language                            |  C++                      | 
| Segment			|  others                         | 

### Description
Hi there,
I have a production system running with a couple of FRAMOS D435e cameras (which is actually the Intel D435 with GigE interface and industrial casing). I have some software stability issues but I don't know if they are caused by the original Intel SDK, the SDK modifications made by Framos or my own implementation. I have been able to reproduce the problem with a D435 USB camera also.

The problem is twofold: 
1. The cameras sometimes stop sending data. Then _poll_for_frames_ and _wait_for_frames_ does not work, even if the software can find the device.
 2. The software freezes in a SDK function call when disconnecting or re-connecting. This happens most frequently in the _pipeline.stop()_ method, but also in _context.query_devices()_ and perhaps some other calls.

I suspect that the problem might be related to the fact that there are two cameras. The implementation is loosely based on the _rs-multicam_ example, but instead of having everything in one big main and a _rs2::pipeline_ vector, I have two instances of a class which has a _rs2::pipeline_ member. The class has a Connect method which uses _context.query_devices()_ to search for the correct camery (by serial number) and calls the _pipeline.start()_ method if it is found.

I haven´t found any actual documentation on how to deal with disconnects, so I don't know if my method is correct, but it works if I test it by manually disconnecting and connecting cameras. It does not work however when the data streaming stops like described above. This is how I do it:
1) if _pipeline.start()_ has been called: call _pipeline.stop()_ (**this usually hangs, if not always**)
2) call my Connect() method:
a) prepare the config:
```
        rs2::config cfg;
        cfg.enable_stream(RS2_STREAM_DEPTH, Par.DepthModule.resWidth, Par.DepthModule.resHeight, RS2_FORMAT_ANY, Par.DepthModule.frameRate);
        cfg.enable_stream(RS2_STREAM_COLOR, Par.RpgModule.resWidth, Par.RpgModule.resHeight, RS2_FORMAT_RGB8, Par.RpgModule.frameRate);

```
b) Try to find the correct camera using the serial number (**sometimes hangs in query_devices()**). _p_ctx_ is a pointer to _rs2::context_ passed to the class constructor such that the two cameras have the same context as in the _rs-multicam_ example. 
```
bool RealSenseCom::GetDevice()
{
    //rs2::context ctx;

    if (Stopping) return false;
    try {
        rs2::device_list devices = p_ctx->query_devices();

        for (rs2::device d : devices) {
            string serial = d.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
            if (Par.Device.SerialNumber.compare(serial) == 0 || (Par.Device.SerialNumber.compare("ANY") == 0)) {
                device = d;
                return true;
            }
        }
        string msg = string("Device with serial number not found\t " + Par.Device.SerialNumber);
        std::cerr << msg << endl;
        LogInfo(msg);
        numConnectionAttempts = 0;
        return false;
    }
    catch (exception & e) {
        std::cerr << "RealSense " << Par.Device.DeviceID << ": GetDevice() exception: " << e.what() << endl;
        string msg = "GetDevice() : exception caught\t " + string(e.what());
        LogInfo(msg);
        return false;
    }    
}
```
c) Start streaming:
```
        cfg.enable_device(device.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
        auto profile = pipe.start(cfg);
```
Note that the software usually runs for hours without problems and sometimes days, but I need it to run infinitely because it is used on a production line.

Thanks.

dorodnic, 2020-04-26:
Hi @EinarBjorn 
Could you please attach call-stack of the hang? If building from source, just hit Pause in the debugger and capture the stack-trace. If using binary SDK from our site, there is an option to install debug symbols with the SDK. With symbols loaded for realsense2.dll, you should be able to see meaningful stack-trace even when not building from source. I don't know how this would work with Framos version, but you mentioned that you can reproduce these problems with regular D435 as well, right?

EinarBjorn, 2020-04-26:
Hi @dorodnic 
Thank you for replying on a Sunday!  :)
The actual production system with the Framos cameras does not have a VS or other debuggers installed. Also, since it is a running production I use another software to monitor and reboot the software when it becomes unresponsive, so it can be tricky to catch the actual hang. Perhaps I can have it create a core dump on exit, but I don't know if the Framos version has the debugging option. I will investigate this.

Worth mentioning: The production system currently has only one camera because the other one is broken. The software attempts to find the missing camera every 1000ms. The system was more stable when bother cameras were operational, although not perfect.

I have very rarely seen the same behavior with the USB camera, where the connected camera stopped streaming and the software hanged in _pipeline.stop()_ method.
However, the software frequently hangs in _query_device()_ on my laptop when I have one USB camera connected but the software expects two cameras and periodically checks for the other one. I will trigger this and post the stack trace.

EinarBjorn, 2020-04-26:
This is all I got. I guess the debug symbols are missing in the Framos version

[image]

EinarBjorn, 2020-04-27:
I managed to build the Framos-librealsense SDK with debug info and dig a little deeper:

[image]

EinarBjorn, 2020-04-27:
### Here is what happens:

- The software expects two cameras and has a pipeline for each one. Only one is connected.
- The main thread tries to find the missing camera every 1000ms
- Worker threads are started by a callback function (triggered by PLC) for each camera every 5-10 seconds (can be quite random):
  - If the camera is connected, the newest frameset from the pipeline is fetched and processed.
  - If the camera is not connected, the worker thread stops immediately

**This can run for hours. Then this occurs:**

1. The main thread hangs when trying to find the missing camera (see image 1).
2. The other pipeline still delivers data for a while (perhaps until the framequeue is empty?). Then there is no data available.
3. A worker thread attempts to disconnect and hangs in pipeline.stop() method (see image 2).
4. Now the software is unresponsive.

[image]

[image]

EinarBjorn, 2020-06-04:
@dorodnic any update on this?

RealSenseSupport, 2020-06-16:
> The implementation is loosely based on the rs-multicam example
I am curious if this can also be reproduced with SDK's rs-multicam with only 1 D435 connected to the host. 
Thank you.

RealSenseSupport, 2020-06-26:
Is this issue still relevant with RealSense D435 using latest SDK and firmware? If so, it may be better to open another issue to focus on multiple RealSense D435 cameras with the most updated sw/fw combo. Please also file the ticket with Framos if this is specific to the sw/fw combination used in the Framos camera.

Please note this ticket will be closed in 7 days if there's no further update. Thank you

EinarBjorn, 2020-06-28:
It is still relevant with the latest Framos SDK and firmware. I don't know how far behind they might be lagging Intel though.
I have also been in contact with their service and provided core dump months ago.

Neither Intel or Framos has commented on this for 2 months since I posted the detailed descriptions and screenshots so I am surprised by this response. It looks like a multi thread problem in the SDK to me because there are several threads waiting for the same mutex. It would be nice to have your opinion on that at least before you close the ticket.

I wonder if it is worth the bother to open another issue.

RealSenseSupport, 2020-08-06:
It seems like the latest Framos software suite, which was released about a month ago, is based on librealsense v2.33.1. Our current SDK is at pre-release v2.37.0 and rs-multicam sample has not yet been able to reproduce the issue.  
It is very crucial to be able to duplicate this issue with one of RealSense family cameras using latest SDK build to allow us to further investigate. Thank you for your understanding. 

Please note this ticket will be closed in 7 days if there's no further update. Thank you

closed 2020-08-13 by RealSenseSupport

lucbettaieb, 2021-06-29:
@RealSenseSupport I find it odd that you auto-closed this issue without resolving it, especially since it affects an industrial product with a company you are partnered with.

EinarBjorn, 2021-06-30:
@lucbettaieb, I agree. The service from their partner Framos  is even worse. I provided detailed information with screenshot, code and core dump but I don't think they even bothered looking into it.