realsenseai/librealsense #6302: Activating CUDA

brsbilgin, 2020-04-26:
| Camera Model                       |  D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra     | 
| Platform                                 |  NVIDIA Jetson   |
| SDK Version                            |  2.32.1             | 

For activating CUDA cores should I build from sources with correct flags or it is enough install from debian packages. How can I evaluate  it is using CUDA cores or not.

MartyG-RealSense, 2020-04-26:
CUDA support can be enabled with a CMake flag called **BUILD_WITH_CUDA**, which is set to False by default.

[image]

CUDA support can be enabled from within a build instruction for the Librealsense SDK using a build instruction similar to the one below:

cmake ../ -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true  -DBUILD_GRAPHICAL_EXAMPLES=true **-DBUILD_WITH_CUDA=true**

An easy way to check whether the effect of CUDA support is being applied is to check CPU usage.  There should be a clear reduction in CPU % usage with CUDa support enabled, as highlighted by the example chart below:

[image]

brsbilgin, 2020-04-26:
Hi MartyG,
Thank you for quick reply. Actually my question is I already install with "sudo apt-get install librealsense2-utils librealsense2-dev -y". For CUDA support should I remove this packages and build according above your instructions. 
Yesterday I build with -DBUILD_WITH_CUDA=true and today with new jetson image I just "sudo apt-get install librealsense2-utils librealsense2-dev -y" I couldn't see any dramatic change between yesterday and today.

dorodnic, 2020-04-26:
Hi @brsbilgin 
Debian packages are already built with CUDA.

brsbilgin, 2020-04-26:
Thank you  Dorodnic

closed 2020-04-26 by brsbilgin