realsenseai/librealsense #6303: Multiple processes, each recording from different camera

timprepscius, 2020-04-26:
Greetings again,

I modified rs-record like so:
```

+    ValueArg<std::string> serialNumber("s", "serialNumber", "blahblahblah", false, "", "");
 
     cmd.add(time);
     cmd.add(out_file);
+    cmd.add(serialNumber);
     cmd.parse(argc, argv);
 
     rs2::pipeline pipe;
     rs2::config cfg;
+
+    if (!serialNumber.getValue().empty())
+        cfg.enable_device(serialNumber.getValue());
+
```

To test recording from different devices in different processes.
It doesn't work.
```
RealSense error calling rs2_pipeline_start_with_config_and_callback_cpp(pipe:0x7fb0bb5527b0, config:0x7fb0bb5562c0, callback:0x7fb0bb554a40):
    failed to set power state

```

Is there a way to make this work?

In a general overview, how would the code need to be modified to make this work? (Running multiple realsense recorders from different processes)

MartyG-RealSense, 2020-04-26:
In 2019, a RealSense user explained how they had  created a multicam record-playback system by combining functions of **rs-multicam** and **rs-record-playback**.

https://github.com/IntelRealSense/librealsense/issues/3802

timprepscius, 2020-04-26:
I have read your rs-multicam

I'm asking whether multiple processes can access different cams.
It seems to crash.

MartyG-RealSense, 2020-04-26:
There was a link to a user's multiple camera recording project in my comment above that explains how they did recording with multiple cameras.

https://github.com/IntelRealSense/librealsense/issues/3802

Their capture and save code to start simultaneous threads is here:

https://github.com/IntelRealSense/librealsense/issues/3802#issuecomment-492992712

timprepscius, 2020-04-26:
Yes, I understand that.

I'm asking about different processes, not threads.

MartyG-RealSense, 2020-04-26:
The documentation link below describes the API rules for multiple applications with multiple cameras.

https://dev.intelrealsense.com/docs/software-support-model#multi-streaming-model

MartyG-RealSense, 2020-05-17:
@timprepscius   Do you still require assistance with this question please, or can the case be closed?   Thanks!

closed 2020-05-24 by MartyG-RealSense