realsenseai/librealsense #6306: Z-accuracy calculation in Depth Quality Tool

panpan0952, 2020-04-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |NA                                | 
| Platform                                 | PC  |
| SDK Version                            |   2.33.1   | 
| Language                            |  python | 
| Segment			|                         | 

### Issue Description
Hi,

I'm writing a python script to get the quality metrics of a D435i following the instruction from Intel® RealSense™ Camera Depth Testing Methodology Rev. 1.0.
I compared the quality metrics calculated by this script and by Quality Tool v2.33.1, then found there was a difference in "z-accuracy".

I set the ground truth to 1000 mm (it's wrong but doesn't matter)
The quality metrics measured by Quality Tool were : Distance: 472.6mm, Angle: 34.48, Z-accuracy:-42.44%.
[image]

The quality metrics measured by my python script were : Distance: 473.88mm, Angle: 34.36, Z-accuracy:-52.61%. I noted that the measured distance was already applied rotation and median in my python script.

In the ch1.3.1 of the methodology document, the z-accuracy was defined as (measured distance - ground truth)/ground truth. So the z-accuracy would be close to (472.6-1000)/1000 = 52.74%. However, it cannot match the z-accuracy measured by Quality Tool, -42.44%.

Is there anything I missed? Thank you!

MartyG-RealSense, 2020-04-27:
For the cased D415 and D435 / D435i cameras, depth is measured from the front of the glass.  So a fixed value is subtracted to get the ground truth.  On the D435 models, '-4.2' mm is added to correct the depth measurement to get the ground truth.

[image]

 As resolution has a bearing on accuracy (and the optimal depth resolution for the D435 models is 848x480, as used by the Depth Quality Tool), is your Python testing application also using 848x480 as its test resolution, please?

ev-mp, 2020-04-27:
@panpan0952 , The `Distance` estimation and the `Z Accuracy` metric do not relate in that way. 
The calculation is done according to 
[image]

Please review the [tool's documentation](https://github.com/IntelRealSense/librealsense/tree/master/tools/depth-quality#z-accuracy) that denotes each component of that equation. You can also use the tool's source code as a reference:  https://github.com/IntelRealSense/librealsense/blob/master/tools/depth-quality/rs-depth-quality.cpp#L105-L139

Additionally, to get a viable characterization the camera shall be positioned strictly perpendicular to the wall. You can use the on-screen cues to orient it properly (tilt <3 Deg).

panpan0952, 2020-04-30:
Thank you MartyG and ev-mp. After modifying my python code, the accuracy can be calculated correctly. 

I understand that the camera should be positioned parallel to the wall. 
The propose of the huge tilt angle is verifying the result of SVD (for plane fitting)

closed 2020-04-30 by panpan0952

MartyG-RealSense, 2020-04-30:
@panpan0952   Thanks so much for the update - I'm happy to hear that you found a solution!