realsenseai/librealsense #6307: D435 depth stream stops when using pipeline

kremerf, 2020-04-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Platform                                 | NUC  |
| Language                            |  C++                          | 

As mentioned in the title I am trying to use a pipeline in my code.

```
cout << "Setting Camera" << endl;
pipeline p;
p.start();
device dev = p.get_active_profile().get_device();
auto sensor = dev.query_sensors()[1];
dev = p.get_active_profile().get_device();
sensor.set_option(RS2_OPTION_EXPOSURE, 100);
cout << sensor.get_info(RS2_CAMERA_INFO_NAME) << endl;
cout << sensor.get_option(RS2_OPTION_EXPOSURE) << endl;
p.stop();
```

But with this the depth stream does not publish any frame. When commenting the lines above out everything works as intended.
Even just including the `p.start()` causes the depth stream to abort.

I am trying to reduce the light received from the RGB image (Exposure set to 1 or alike), but I'd like to use the depth camera in its default state. They do not need to be linked together, I can perform that translation myself, if need be.

The output from the lines above, when used, is:
(In this snipped the set_option line was commented out, so the default value is shown.
The snippet is called right after the reset of the D435, to actually write said values into the config, but before it starts working - as I just get DEVICE BUSY errors when performing this task later on in the code.

```
[ INFO] [1587980176.928873721]: Resetting device...
Setting Camera
RGB Camera
166
```

Additional Information:
I am using several ROS-Nodes to listen to the camera topics which then use that data and send it further. Currently I am just trying to get all the data upon startup. As the data-stream fixes itself, when unplugging and replugging the D435. But that is not viable in this project, so I need to get this fixed.

MartyG-RealSense, 2020-04-27:
Please try replacing **pipeline p** with this:

**rs2::pipeline p;**

kremerf, 2020-04-27:
Forgot to mention, that I have `using namespace rs2;` in my code. Tried it either way, and it does not seem to change anything.

MartyG-RealSense, 2020-04-27:
There is a reference script here that may provide some useful guidance to check your own code structure against:

https://github.com/IntelRealSense/librealsense/issues/1624#issuecomment-389085387

kremerf, 2020-04-27:
I feel like I got most of the lines used from that one^^
Unfortunately the code used there does generate `Device Busy` and `rs2::backend_error`

MartyG-RealSense, 2020-04-27:
The SDK documentation describes the meaning of **rs2::backend_error** as a "Failure returned from system call".  

If an error such as Device Busy occurs, a workaround may be to add an error "catch" where the program keeps trying until it gets past the error, like the example code here:

https://github.com/IntelRealSense/librealsense/issues/2240#issuecomment-413555585

kremerf, 2020-04-27:
With that catch we can see the following:

```
 27/04 13:08:16,369 ERROR [139878871598848] (types.h:304) xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
[ERROR] [1587985696.369224246]: An exception has been thrown: xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
terminate called after throwing an instance of 'rs2::backend_error'
  what():  xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
[D435/realsense2_camera_manager-1] process has died [pid 22627, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/paytah/.ros/log/6a5049bc-8852-11ea-8a7d-1856805a244a/D435-realsense2_camera_manager-1.log].
log file: /home/paytah/.ros/log/6a5049bc-8852-11ea-8a7d-1856805a244a/D435-realsense2_camera_manager-1*.log
exposure = 166
meta exposure = RealSense error calling rs2_get_frame_metadata(frame:0x55dc8d1fc2e8, frame_metadata:Actual Exposure):
    metadata not available
[detector_processing-3] process has died [pid 22629, exit code 1, cmd /home/paytah/catkin_ws/devel/lib/detector/detector_processing __name:=detector_processing __log:=/home/paytah/.ros/log/6a5049bc-8852-11ea-8a7d-1856805a244a/detector_processing-3.log].
log file: /home/paytah/.ros/log/6a5049bc-8852-11ea-8a7d-1856805a244a/detector_processing-3*.log
[D435/realsense2_camera-2] process has finished cleanly

```

kremerf, 2020-04-27:
This is my current main.cpp (Some parts removed, that don't have anything to do with the realsense camera.)

```
#include <librealsense2/rs.hpp>
#include <chrono>
#include <thread>
using std::cout;
using namespace rs2;

int main(int argc, char** argv) try
{
    ros::init(argc, argv, "detector_processing");
    ROS_INFO("Started >Processing<-ROS-Node");
    while(ros::ok() ){
		if (/*Right after Reset*/) {
			cout << "Setting Camera!" << endl;
			// rs2::pipeline p;
			// p.start();
			// device dev = p.get_active_profile().get_device();
			// auto sensor = dev.query_sensors()[1];
			// dev = p.get_active_profile().get_device();
			// sensor.set_option(RS2_OPTION_EXPOSURE, 250);
			// cout << sensor.get_info(RS2_CAMERA_INFO_NAME) << endl;
			// cout << sensor.get_option(RS2_OPTION_EXPOSURE) << endl;
			// p.stop();
			rs2::pipeline pipe;
			pipe.start();
			rs2::device dev = pipe.get_active_profile().get_device();

			dev.query_sensors()[0].set_option(rs2_option::RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
			std::this_thread::sleep_for(std::chrono::seconds(1));
			std::cout << "1exposure = " << dev.query_sensors()[0].get_option(rs2_option::RS2_OPTION_EXPOSURE) << std::endl;

			auto frameset = pipe.wait_for_frames();
			std::cout << "2meta exposure = " << frameset.get_depth_frame().get_frame_metadata(rs2_frame_metadata_value::RS2_FRAME_METADATA_ACTUAL_EXPOSURE) << std::endl;

			dev.query_sensors()[0].set_option(rs2_option::RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);
			std::this_thread::sleep_for(std::chrono::seconds(1));
			std::cout << "3exposure = " << dev.query_sensors()[0].get_option(rs2_option::RS2_OPTION_EXPOSURE) << std::endl;

			frameset = pipe.wait_for_frames();
			std::cout << "4meta exposure = " << frameset.get_depth_frame().get_frame_metadata(rs2_frame_metadata_value::RS2_FRAME_METADATA_ACTUAL_EXPOSURE) << std::endl;
		}
        ros::spinOnce();
        rate.sleep();
    }

    ros::spin();

    return 0;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception & e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}

```

MartyG-RealSense, 2020-04-27:
Metadata is supported by default but on Linux the metadata support needs to be patched in when building from source using the usual Librealsense installation process.

https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md#os-support

The easiest way to add metadata support on Linux is to build Librealsense using a 'bypass' method. 
 It should work with any type of Linux and is not dependent on kernel versions or patches, so should help to eliminate the possibility that the problem is being caused by the Linux kernel.  An internet connection is required for this method of installation.

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

kremerf, 2020-04-27:
I might be on to something - 

```
{
			cout << "Setting Camera!" << endl;
			rs2::pipeline p;
			p.start();
			device dev1 = p.get_active_profile().get_device();
			auto sensor = dev1.query_sensors()[1];
			dev1 = p.get_active_profile().get_device();
			sensor.set_option(RS2_OPTION_EXPOSURE, 1);
			cout << sensor.get_info(RS2_CAMERA_INFO_NAME) << endl;
			cout << sensor.get_option(RS2_OPTION_EXPOSURE) << endl;
			p.stop();
			rs2::pipeline pipe;
			pipe.start();
			rs2::device dev = pipe.get_active_profile().get_device();

			dev.query_sensors()[0].set_option(rs2_option::RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
			// // std::this_thread::sleep_for(std::chrono::seconds(1));
			// std::cout << "1exposure = " << dev.query_sensors()[0].get_option(rs2_option::RS2_OPTION_EXPOSURE) << std::endl;
			//
			auto frameset = pipe.wait_for_frames();
			// std::cout << "2meta exposure = " << frameset.get_depth_frame().get_frame_metadata(rs2_frame_metadata_value::RS2_FRAME_METADATA_ACTUAL_EXPOSURE) << std::endl;
			//
			// dev.query_sensors()[0].set_option(rs2_option::RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);
			// // std::this_thread::sleep_for(std::chrono::seconds(1));
			// std::cout << "3exposure = " << dev.query_sensors()[0].get_option(rs2_option::RS2_OPTION_EXPOSURE) << std::endl;
			//
			// frameset = pipe.wait_for_frames();
			// std::cout << "4meta exposure = " << frameset.get_depth_frame().get_frame_metadata(rs2_frame_metadata_value::RS2_FRAME_METADATA_ACTUAL_EXPOSURE) << std::endl;
		}
```

This works, without the cmake installation, I tried this just until now. So basically the command `wait_for_frames()` seems to solve it. Which makes sense from its name, as it is waiting for the frames I want to see.. but I don't know what part it actually fixed internally^^

MartyG-RealSense, 2020-04-27:
Glad you made progress!

What wait_for_frames does is that it blocks until a frame is available.

wait_for_frames and alternative frame control instructions such as poll_for_frames are explained here:

https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

kremerf, 2020-04-27:
I will look into that, i copied the code from above again into my program and it worked. I will have a look at this and at #2422. If any further problems occur I will reopen this.

Thank you very much for your help, once again!

closed 2020-04-27 by kremerf