realsenseai/librealsense #6309: Can't start depth image capture using any of provided scripts

1a11, 2020-04-27:
| | |
|---|---|
|**librealsense**|2.33.1 RELEASE|
|**OS**|Windows 10|
|**CPU**|AMD Ryzen 5 1500x|
|**Name**|Intel RealSense SR300|
|**Serial Number**|619206000621|
|**Firmware Version**|3.21.0.0|
|**Camera Locked**|YES|
|**Product Line**|SR300|
|**Asic Serial Number**|619206000621|
|**Firmware Update Id**|619206000621|
|**Using USB3.1**|YES|

Can't start depth image capture using any of provided scripts. Also got some errors when starting:

`
Backend in rs2_get_option_range(options:0000013F11F718E0, option:Confidence Threshold, min:-nan, max:0, step:4.47014e-43, def:3.96723e-28):
ks_control->KsProperty( reinterpret_cast<PKSPROPERTY>(&node), sizeof(node), &description, sizeof(KSPROPERTY_DESCRIPTION), &bytes_received) returned: HResult 0x80004002: "Интерфейс не поддерживается"
`

Here is the log I saved for debugging:

` 27/04 14:52:37,709 INFO [1896] (rs.cpp:2283) Framebuffer size changed to 958 x 1018
 27/04 14:52:37,709 INFO [1896] (rs.cpp:2283) Window size changed to 958 x 1018
 27/04 14:52:37,709 INFO [1896] (rs.cpp:2283) Scale Factor is now 1
 27/04 14:52:37,797 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:37,813 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:37,827 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:37,864 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:37,888 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:37,889 INFO [6484] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Представлен недопустимый номер потока."
 27/04 14:52:37,889 INFO [6484] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Представлен недопустимый номер потока."
 27/04 14:52:37,889 INFO [6484] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Представлен недопустимый номер потока."
 27/04 14:52:37,889 INFO [6484] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Представлен недопустимый номер потока."
 27/04 14:52:37,898 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:37,906 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:37,915 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:37,923 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:37,932 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:37,940 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:37,948 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:37,957 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:37,963 INFO [1896] (win-helpers.cpp:87) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Элемент не найден."
 27/04 14:52:37,965 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:37,974 INFO [6484] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:38,000 INFO [6484] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Представлен недопустимый номер потока."
 27/04 14:52:38,000 INFO [6484] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Представлен недопустимый номер потока."
 27/04 14:52:38,000 INFO [6484] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Представлен недопустимый номер потока."
 27/04 14:52:38,000 INFO [6484] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Представлен недопустимый номер потока."
 27/04 14:52:39,713 INFO [1896] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:39,722 INFO [1896] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:39,746 INFO [1896] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:39,754 INFO [1896] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:39,780 INFO [1896] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:39,788 INFO [1896] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:39,813 INFO [1896] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:39,821 INFO [1896] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:39,847 INFO [1896] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:39,855 INFO [1896] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:42,469 INFO [1896] (sensor.cpp:1338) Request: Z16 Depth, 
Resolved to: INZI Depth, 
 27/04 14:52:42,478 INFO [1896] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:42,873 INFO [2548] (win-helpers.cpp:87) owner->_reader->ReadSample(dwStreamIndex, 0, nullptr, nullptr, nullptr, nullptr) returned: HResult 0xc00d3ea2: "Устройства видеозаписи больше нет."
 27/04 14:52:42,877 WARNING [2320] (sr300.cpp:85) SR300 group_devices is empty.
 27/04 14:52:42,878 DEBUG [2320] (context.cpp:384) 
Device disconnected:

uvc devices:
id- 
[... 73 lines cut ...]
[... cut ...]
 27/04 14:52:44,038 INFO [1896] (win-helpers.cpp:87) _source->QueryInterface(__uuidof(IAMVideoProcAmp), reinterpret_cast<void **>(&_video_proc)) returned: HResult 0x80004002: "Интерфейс не поддерживается"
 27/04 14:52:46,365 INFO [1896] (synthetic-stream-gl.cpp:96) Shutting down rendering
 27/04 14:52:46,365 INFO [1896] (synthetic-stream-gl.cpp:102)  0 GPU objects cleaned-up
 27/04 14:52:46,365 INFO [1896] (synthetic-stream-gl.cpp:152) Shutting down processing
 27/04 14:52:46,366 INFO [1896] (synthetic-stream-gl.cpp:163)  0 GPU objects cleaned-up
`

The only way it works is when the laser projector is turned off.

MartyG-RealSense, 2020-04-27:
Please try using the RealSense Viewer program, setting its 'Laser Power' setting to '0' under the "Controls" section of the Viewer's side-panel options, and then activate a depth stream.  

If the stream works, please try increasing the Laser Power value above '0' in small increments until the camera disconnects.

If the camera initially streams when Laser Power = 0 and then disconnects when Laser Power exceeds a certain value, this will indicate that there is potentially an issue with the camera not being supplied with enough power to keep its connection stable.

dorodnic, 2020-04-28:
It seems that the host PC is not able to match power requirement of the device. Insufficient power will cause the device to disconnect. SR300 has measurable current increase when starting streaming, causing the problem. 
If you are using a laptop, please make sure it is connected to the wall. If you are using a hub, please consider powered USB hub.

1a11, 2020-04-28:
@dorodnic Will buy. Thanks for the help. Consider this thread closed, please.

SUMMARY: If you can't properly start SR300 depth info stream, consider buying powered USB hub or just use Intel PC cos it seems it's much simple that way.

closed 2020-04-28 by 1a11