realsenseai/librealsense #6311: Left border area of D415 right IR stream corrupted

robertxwu, 2020-04-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 / D415 and D435           | 
| Firmware Version                   | 05.12.03.00                              | 
| Operating System & Version | Win 10                                     | 
| Kernel Version (Linux Only)    | N/A                                         | 
| Platform                                 | PC                                           |
| SDK Version                           |  v2.32.1, v2.33.1, v2.34.0          | 
| Language                               |  N/A                              | 
| Segment			         |  Robot                          | 

### Issue Description
Left border area of right IR stream is often corrupted, as shown in the image below. This happens on my both D415 and D435.

All you assistance will be greatly appreciated!

[image]

robertxwu, 2020-04-27:
It looks like this happens when Disparity Shift is not zero. Since my work distance is around 200 millimeters, I have found that setting Disparity Shift to 128 produces reasonable good depths. 

Given the work distance, could you please let me know if there is a way to get both decent depths and non-corrupted IR images? Thanks

MartyG-RealSense, 2020-04-27:
There has been a previous case where this edge effect was reported to have occurred with the right IR frame.

https://github.com/IntelRealSense/librealsense/issues/2480

The edge distortion is slightly hard to see in the original image in that link, so I have enhanced the image to highlight the pattern.

[image]

robertxwu, 2020-04-27:
Here is an example of depth stream with IR streams
[image]

MartyG-RealSense, 2020-04-27:
If I had seen the depth image before the Ir images then my first instinct may have been that it was an effect called an 'Invalid Depth Band'.  The current edition of the data sheet document for the 400 Series cameras describes in on pages 59 and 60.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

The data sheet's opening paragraph of the Invalid Depth Band section describes the phenomenon like this:

"The depth data generated with stereo vision uses the left imager as the reference for
stereo matching resulting in a non-overlap region in the field of view of left and right
imagers where we will not have depth data at the left edge of the frame.  Closer
scenes result in a wider invalid depth band than scenes at larger distances".

Outside of your case and the other one that I linked to though, I have not seen the line pattern effect on the edge in cases of Invalid Depth Band, which makes me think that it might be both an Invalid Depth Band and some other effect.

robertxwu, 2020-04-28:
@MartyG-RealSense always appreciate your prompt & great response. I can confirm the width of the corrupted area is equal to the disparity shift value.

-

dorodnic, 2020-04-28:
I think this is expected effect of applying disparity shift (it is literally shifting the IR image to get more disparity values after stereo matching). @agrunnet - do you know if this is to be expected?

robertxwu, 2020-04-28:
A even more strange finding: After I set the disparity shift to a non-zero value, to say 128, I had to deduct disparity by the shift in order to get correct depth using formula stated at:
https://docs.opencv.org/master/dd/d53/tutorial_py_depthmap.html

If so, why need to shift the right stream? This not only creates unwanted visual effect and also brings confusion for people using left-right IR streams as standard stereo camera. 

Could we not shift the right stream, instead shifting x coordinates of correspondences detected on the right image?

agrunnet, 2020-04-29:
Sorry, this is how the disparity shift is implemented. The right image is shifted (but not the left which is kept as the reference).  
I did just test and confirmed on D435 that as you change the disparity shift, the depth value doe not change, meaning the Z value stays unchanged.  I am not sure why you need to deduct disparity to get correct depth.

robertxwu, 2020-04-29:
@agrunnet thank you for taking time to verify like that. However, my use case is different, I use the RealSense D415 as a standard stereo camera: acquire the left/right IR images, detect feature correspondences on the left/right images on my PC with OpenCV, and then triangulate the correspondences using intrinsic & extrinsic retrieved from the D415. In addition, I want to verify my correspondence matching and triangulation results with depth information directly from the camera. Since my work distance is about 200mm, I had to set the disparity shift to 128, in order to get decent depth info from the camera. 

Then, I found out that the disparities computed on my PC side must be deducted by the disparity shift value to get correct triangulation results.

If my finding is right, is there a way to get the right IR stream without being shifted when the disparity shift value > 0?

agrunnet, 2020-04-29:
1. Why do you not just use the depth calculation done by the asic?
2. If you use your own algorithm you can do whatever you want to the right and left images, meaning there is not need to do a disparity shift on the ASIC. Or am I missing something?

For near range application I would recommend the D435. The image is in focus a lot closer. The D415 starts to blur at about 40cm.

robertxwu, 2020-04-29:
@agrunnet Our current software uses standard stereo cameras for stereoscopic vision, we are trying to use RealSense D415 to replace them, and want to sanity check our triangulation results with the depth done on the ASIC.

For D435, I evaluated it for our application, and found out that pixel resolution (mm per pixel) at our work distance was not good enough.

MartyG-RealSense, 2020-05-17:
Hi @robertxwu   Do you require further assistance with this case please, or can the case be closed?  Thanks!

robertxwu, 2020-05-17:
@MartyG-RealSense I am OK to have this closed. I understand this is more a feature request: I would have an option to download non-shift IR image even Disparity Shift > 0.
Thank you all very much for your helps/comments.

MartyG-RealSense, 2020-05-17:
@robertxwu   To make your request easy to find, given the length of this discussion, please create a new issue just for your request, and I will close this discussion and label your Feature Request when it is created.  Thanks!.

closed 2020-05-17 by MartyG-RealSense