realsenseai/librealsense #6312: Accessing pointer to frame metadata

mpsilver, 2020-04-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version | Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    | 5.3.0-46 | 
| Platform                                 | PC |
| SDK Version                            | 2.34.0 | 
| Language                            | C++ | 

### Issue Description
I am currently streaming depth data from the D435 camera and saving this data to an HDF5 file. For each frame, I would also like to save all available metadata fields, but I want to avoid making individual frame.get_frame_metadata() calls for each metadata value. Is there a way that I could get a pointer to the start of the frame metadata, similar to how frame.get_data() returns a pointer to the frame data? Looking through the source code, this might be equivalent to accessing a pointer to frame.additional_data, but I am not exactly sure how to do this.

dorodnic, 2020-04-28:
Hi @mpsilver 
There is no such API currently, but iterating over supported metadata enum values should not affect performance (should be negligible). Calls to `rs2_get_frame_metadata` do not perform any IO and are just returning values from internal `frame.additional_data` struct.

RealSenseSupport, 2020-05-04:
Hi @mpsilver Do you need further help on this?
We will close this issue if we don't hear from you in 7 days.

closed 2020-05-14 by RealSenseSupport