realsenseai/librealsense #6313: less frames publish after filter in pyrealsense2

MikoyChinese, 2020-04-28:
When I run my D435 with (848, 480) resolution in depth and color, I can get more than 30fps(max: 49fps) in my device, but if I try to turn on filter such as `spatial`, `temporal` filter in my code, I just can get ~20 fps in my devices. And I check my cpu, it has only 30%  usage while after filter is up to 40-50% usage.

So who can tell me why can't I get 30fps in my device? How to solve it? Thx.

Best,
Mikoy Chinese

MartyG-RealSense, 2020-04-28:
Applying post-processing filters has an additional processing overhead.  The link below has advice for trying to enforce a constant FPS.

https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579

An alternative way to improve performance is to offload work from the CPU to the graphics GPU if your computing device has a good GPU.

https://github.com/IntelRealSense/librealsense/issues/4905#issuecomment-533854888

dorodnic, 2020-04-28:
Hi @MikoyChinese 
Could you please share platform details? 
If you are building SDK from source please make sure you enable `-DCMAKE_BUILD_TYPE=Release` flag

closed 2020-05-17 by MartyG-RealSense