realsenseai/librealsense #6316: RealeSense T265 keeps connecting and disconnecting

oaphek, 2020-04-28:
The symptoms:

The device worked well in the beginning. It gradually deteriorated.

With the Realsense Viewer on windows 10, after turning on with the toggle from red to green, it works OK and then after a few seconds you hear the windows USB disconnect - connect sounds, and the Viewer toggle goes back to red.

On Linux Ubuntu where I use it as a developer, I get the Wait error:
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x8ceea0):
Frame didn't arrived within 5000

This happens in Windows and Realsense, with the original cable and an alternative able, in all USB ports, with and without other devices connected.

Many thanks,

Ori Aphek

| | |
|---|---|
|**librealsense**|2.31.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense T265|
|**Serial Number**|908412111215|
|**Firmware Version**|0.2.0.879|
|**Product Line**|T200|

Please provide a description of the problem

m4xr1sk, 2020-06-18:
My first T265 camera does the same.
Solved with the trashcan, no support from Intel

oaphek, 2020-06-18:
Hi,

Eventually, I did get hold of Intel's support. It took a while to find how
to do it. I had my unit replaced. The new one works fine for now.

Ori Aphek

Founder and CTO

(M) +972 52 8118601

(US) 303 396 3667
ori.aphek@siteaware.com
www.siteaware.com

<https://www.linkedin.com/in/ori-aphek-4705/>

This email and any attachments may contain private, confidential and
privileged material for the sole use of the intended recipient. If you are
not the intended recipient, please immediately delete this email and any
attachments

closed 2021-03-18 by RealSenseSupport