realsenseai/librealsense #6319: realsense is always damaged and cannot recognize usb

xiaoniu001, 2020-04-29:
I connected realsense to Raspberry Pi 4+, and i can run for a few days,  but it is always damaged and i can't find reason.Please someone tell me why

MartyG-RealSense, 2020-04-29:
If you are referring to doing a long-running session with the camera that lasts for days ... the 400 Series cameras can run indefinitely so long as their temperature remains within the official maximum recommended operating temperature of 35 degrees C (42 degrees C may be the practical real-world limit before problems start occurring).

If the camera temperature remains within tolerances then when problems with long-running sessions occur, they are typically related to a problem that is occurring outside of the camera hardware.  This could include a glitch on the USB port / USB cable or a problem with the computer hardware or its operating system.  Resetting the camera will often correct the problem.

In situations where it is not practical to manually reset the camera (because it must be automated or because the computer is locked away in a cabinet), it is possible to program an instruction called **hardware_reset()** in a script to auto-reset the camera.  It can be set up to reset if certain conditions are detected, such as more than 50% of the pixels in a depth image having a depth value of zero.

xiaoniu001, 2020-04-30:
@MartyG-RealSense First of all thank you for your reply，I have damaged two cameras, they are both running for a few days and can no longer recognize usb. So the biggest possibility is that the running time is too long and the camera temperature is too high, right?

MartyG-RealSense, 2020-04-30:
If the cameras can work for a few days before they are no longer recognised by USB then they are not damaged, and something else is causing the problem.  The cameras have built-,in safety shutdown for the laser if their detected temperature goes higher than 60 degrees for longer than several seconds so it would be hard for temperature to permanently damage them.

The cameras are intended to be able to run for an unlimited time period if their temperature remains under a certain level.   So the problem is not that you are running them for too.long, but that something is probably going wrong on the computer that the camera is attached to.

3 to 4 days is commonly when problems start occurring if they are going to happen.  Using "industrial grade" computer and USB equipment may help to increase stability.  For example, industrial grade USB cables are designed for critical applications such as medical scanners and nuclear reactors, where the dropping of frames cannot be permitted. 

Do you have 2 cameras attached to one Pi 4 please?

xiaoniu001, 2020-05-06:
> If the cameras can work for a few days before they are no longer recognised by USB then they are not damaged, and something else is causing the problem. The cameras have built-,in safety shutdown for the laser if their detected temperature goes higher than 60 degrees for longer than several seconds so it would be hard for temperature to permanently damage them.
> 
> The cameras are intended to be able to run for an unlimited time period if their temperature remains under a certain level. So the problem is not that you are running them for too.long, but that something is probably going wrong on the computer or the USB cable / port that the camera is attached to.
> 
> 3 to 4 days is commonly when problems start occurring if they are going to happen. Using "industrial grade" USB equipment may help to increase stability. For example, industrial grade USB cables are designed for critical applications such as medical scanners and nuclear reactors, where the dropping of frames cannot be permitted.
> 
> Do you have 2 cameras attached to one Pi 4 please?

Thank you for your answer.  I  only connected one camera to Pi

MartyG-RealSense, 2020-05-06:
If you touch the camera casing a few minutes after you start the program and you find that the casing is already hot, that could indicate there is a bad USB cable or a fault on the USB port.

External air temperature in the environment can also affect internal camera operating temperature.  The latest guidance about maximum recommended outer case temperature in the May 2020 edition of the data sheet document for the 400 Series cameras is that the case temperature should not exceed 50 degrees C.  Lab tests by Intel found that if the case temperature reached 60 degrees then that could translate to an internal operating temperature of 45 degrees C.

xiaoniu001, 2020-05-07:
> If you touch the camera casing a few minutes after you start the program and you find that the casing is already hot, that could indicate there is a bad USB cable or a fault on the USB port.
> 
> External air temperature in the environment can also affect internal camera operating temperature. The latest guidance about maximum recommended outer case temperature in the May 2020 edition of the data sheet document for the 400 Series cameras is that the case temperature should not exceed 50 degrees C.
> 
> Lab tests by Intel found that if the case temperature is below 44 degrees C then the internal laser projector temperature should not exceed 60 degrees C (the point where the laser safety power reduction / shut-off mechanism may activate).
> 

> [#5209 (comment)](https://github.com/IntelRealSense/librealsense/issues/5209#issuecomment-563358823)

Thanks a lot for your professional reply.
Now the situation is the Linux PC can not recognized the device, which "lsusb" output is nothing.
According to your reply, may be the safety power reduction is now in active status, I have tried a lot of things to inactive the status, but it doesn't work.
I wonder if you could give me some advice to inactive the safety protection so that I can continue to use the camera.

MartyG-RealSense, 2020-05-07:
There is nothing that can be done to override the camera's laser safety function.  It is necessary to try to stop the camera from reaching that 60 degrees C laser shutdown temperature.

If the problem is with the Pi 4 passing temperature to the camera by heating up the USB cable and the end connector that plugs into the side of the camera, an inexpensive heatsink can be added to the Pi board to extract more of the heat away from the board.  Here is an example product:

https://thepihut.com/products/xl-raspberry-pi-4-heatsink

If the USB cable is getting very hot but the Pi board is cool, this could be an indicator of a quality problem with the cable.  If the cable is hot and you have a spare USB-C cable, try the alternative cable and see how this affects camera temperature.

If the board and cable are cool but the camera temperature is high, you could try using a lower camera Laser Power value.  This setting can be controlled in the RealSense Viewer control panel or with scripting.  A drawback of a lower Laser Power setting is that the depth detail on an image may become more sparse.

closed 2020-05-11 by xiaoniu001