realsenseai/librealsense #6320: How to achieve Self-Calibration  in C/C++

dingbaiyin, 2020-04-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Linux (Ubuntu 16/) / | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<3>.<4> }                          | 
| Language                            |  {C、C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
has been stuck in these two days, no progress.Seeking expert guidance

dingbaiyin, 2020-04-29:
This is difficult for me, if you have examples, I hope you can refer to your code
Thank you

MartyG-RealSense, 2020-04-29:
A reply was posted at the link below.  Thanks!

https://github.com/IntelRealSense/librealsense/issues/6231#issuecomment-621013375

MartyG-RealSense, 2020-05-17:
Hi @dingbaiyin   Do you require further assistance with this question please, or can the case be closed?  Thanks!

closed 2020-05-24 by MartyG-RealSense