realsenseai/librealsense #6322: modprobe: FATAL: Module videodev is in use. Failed to unload module videodev. error type 1. Operation is abort.

AlessandroMelino, 2020-04-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |                             D400                           | 
| Firmware Version                   |                             Latest                          | 
| Operating System & Version |          Ubuntu Server 18.04 LTS               | 
| Kernel Version (Linux Only)    |                        5.3.0                                | 
| Platform                                 |                  Raspberry Pi 4                       |
| SDK Version                           |                       2.34.0                               | 
| Language                              |  {C/C#/labview/nodejs/opencv/pcl/python/unity }  | 
| Segment			        |                          Robot                       | 

### Issue Description

Hello everyone.

As you can see I have a Raspberry PI 4 with Ubuntu Server 18.04 and a desktop installed. I am trying to install librealsense to manage my D435 camera but in the step of the [installation tutorial](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)

>  Prepare Linux Backend and the Dev. Environment

And execute the fourth command: 

`./scripts/patch-realsense-ubuntu-lts.sh
`

After some minutes I get the following error message:

> modprobe: FATAL: Module videodev is in use.
> Failed to unload module videodev. error type 1. Operation is abort.

Reading some issues, it can be about the kernel version (I have 5.3.0 but in the last realease of librealsense it is supported) or about havin the Secure Boot ON.

I tried to turn off the secure boot but the next error appear:

> EFIvariables are not supported on this system

So I don't know how to solve the problem and instal librealsense on my Raspberry PI4.

Thanks in advance.

Best regrets.
Alessandro

MartyG-RealSense, 2020-04-29:
If you are having difficulties with a patching-related part of the usual installation process then there is a "bypass" installation option that is not dependent on Linux versions, patches or kernel versions.  It requires an internet connection to install.

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

AlessandroMelino, 2020-04-29:
> If you are having difficulties with a patching-related part of the usual installation process then there is a "bypass" installation option that is not dependent on Linux versions, patches or kernel versions. It requires an internet connection to install. A typical example of a CMake build instruction for the bypass would look like this:
> 
> **cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

Thanks for the quick answer @MartyG-RealSense .

So I have to execute this in the root directory of librealsense that I cloned from github? Or what.

And then, if everything works fine what is the next step of the installation guide I must follow?

Best regrets.

MartyG-RealSense, 2020-04-29:
Don't worry about being new.  Everyone was new once.  :)

It should be executed in the **Build** folder of the Librealsense root (Librealsense > Build).

As far as I know, once the instruction is run then there are no more installation steps after that.

AlessandroMelino, 2020-04-29:
Finally it works. I am so grateful for your help @MartyG-RealSense .

By the way, when I run the realsense viewer, appears an error that the ms() take too long or something like that. I saw others posts with this issue and you said that it was waiting for a fix. Is it already in that state? Or exist a fix for this error?

Best regrets.

MartyG-RealSense, 2020-04-29:
Great news that it worked for you!   :)

It looks as though the fix in the pull is still awaiting review.  

https://github.com/IntelRealSense/librealsense/pull/6150

One RealSense user who did not want to wait for the merge said that they would try to use the code listed on the pull page to insert it into their own custom compilation of the SDK.  For most people though, the options are to use SDK 2.33.1 (the previous version) or wait for the fix to be implemented.  I have found that the Viewer in SDK 2.34.0 is still usable though whilst the error is being generated, though there is a freeze problem with the On-Chip Calibration option.

AlessandroMelino, 2020-04-29:
Thank you for your answer, for my application is just enough to check that the camera works, so I will wait for the official fix. Every doubt was solved so I close the issue.

Best regrets.

closed 2020-04-29 by AlessandroMelino

hustchenli, 2020-05-17:
> Don't worry about being new. Everyone was new once. :)
> 
> It should be executed in the **Build** folder of the Librealsense root (Librealsense > Build).
> 
> As far as I know, once the instruction is run and it has completed successfully then there are no more installation steps after that.

Hello, I just excuted this command. But there is no **Build** folder in librealsense. I don't know why. Could you please give me some suggestions?
Best regards

MartyG-RealSense, 2020-05-17:
My understanding is that the **Build** folder is involved if you are building Librealsense from source code on Linux using the instructions below:

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

The **Build** folder is not present by default.  You have to create it in the Librealsense root directory and then change to the newly created Build directory using this instruction:

**mkdir build && cd build**

If you are a Windows user, you can instead set the customisation settings with tick-boxes in a visual interface instead by running the **CMake** program.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

hustchenli, 2020-05-17:
> The **Build** folder is required if you are building Librealsense from source code on Linux using the instructions below:
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
> 
> The **Build** folder is not present by default. You have to create it in the Librealsense root directory and then change to the newly created Build directory using this instruction:
> 
> **mkdir build && cd build**
> 
> Then you input the CMake build instruction whilst you are in the Build folder to begin building Librealsense with the conditions that you set in your CMake instruction..
> 
> If you are a Windows user, you can instead set the CMake customisation settings with tick-boxes in a visual interface instead by running the **CMake** program once it has been downloaded and installed.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md
> 
> The easiest way for Windows users to install the RealSense SDK though is to install the pre-built version available from the 'Assets' list at the bottom of each version's details on the SDK **Releases** page:
> 
> https://github.com/IntelRealSense/librealsense/releases/
> 
> [image]

Thanks for your answer. I'm Linux user(Ubuntu 18.04). I created build folder in librealsense using this command: 
**mkdir build && cd build**
and then excuted:
**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**
then excuted in librealsense:
**./scripts/patch-realsense-ubuntu-lts.sh**.
But I still get this result:
**Patched kernels modules were created successfully

modprobe: FATAL: Module videodev is in use.
Failed to unload module videodev. error type 1 . Operation is aborted**

MartyG-RealSense, 2020-05-17:
If you are using this -DFORCE RUSB method of installation, this is the backend "bypass" method that uses an internet connection and does not rely on Linux versions, kernel versions or patches.  You therefore do not need to run the patch instruction after the install.  Using this installation method, there should be no further steps to do and the build of Librealsense should have completed successfully.

hustchenli, 2020-05-17:
@MartyG-RealSense Do you mean that there is no need to excute:
**./scripts/patch-realsense-ubuntu-lts.sh.** ?

MartyG-RealSense, 2020-05-17:
That's right, the internet connection installs all that you need, avoiding the need to patch it.

MartyG-RealSense, 2020-05-17:
The link below has an explanation from a RealSense team member of the advantages and disadvantages of using the no-patch method versus using patching, in the section of the comment headed '**What are the advantages and disadvantages of using libuvc vs patched kernel modules?**'

https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604

hustchenli, 2020-05-17:
@MartyG-RealSense Thanks a lot for your explanation and help.