realsenseai/librealsense #6323: deproject pixel to point with color information

fburak, 2020-04-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Platform                                 | PC  |
| Language                            |  {C++/pcl }                          | 

Hello,

I successfully align rs2::depth_frame and rs2::video_frame objects and deproject some pixels (that refers to some area in color image taken by realsense d435i camera) to 3D points. After that i push back these 3D points to a pcl::PointCloud\<pcl::PointXYZ\> object in order to get a point cloud representation of part of image. However i would like to generate this point cloud with color information in format of pcl::PointCloud\<pcl::PointXYZRGB\>. I searched a lot but cannot find answer to my question? Any solutions in librealsense? How can i achieve this? Any help will be appreciated.

Thanks

MartyG-RealSense, 2020-04-29:
I hope that the link below, that contains some scripts contributed by RealSense community members, will be of help to you.

https://github.com/IntelRealSense/librealsense/issues/1601

[image]

MartyG-RealSense, 2020-05-17:
Hi @fburak   Do you require further assistance with this question please, or can the case be closed?  Thanks!

fburak, 2020-05-20:
Hello @MartyG-RealSense , 

Problem is solved thank you. 

For those who have the same problem, it can be achieved by pushing r, g and b data that comes from color image along with the x, y, and z coordinate data that comes from deproject_pixel_to_point function to the pcl::PointXYZRGB object manually and generating pcl::PointCloud<pcl::PointXYZRGB> point cloud object from pcl::PointXYZRGB objects.

closed 2020-05-20 by fburak

MartyG-RealSense, 2020-05-20:
Great news - thanks so much for sharing your method with the RealSense community!