realsenseai/librealsense #6324: rs2::points Class Reference get_vertices documentation

PerceptMD, 2020-04-29:
Just a small thing but maybe you would like to update this.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1points.html

> 
Parameters:
[in] | vertex* | - pointer of vertex sturcture
-- | -- | --

the vertex is an out param because it is returned.

```
        /**
        * Retrieve the vertices of the point cloud
    ->> * \return vertex* - pointer of vertex sturcture
        */
        const vertex* get_vertices() const
        {
            rs2_error* e = nullptr;
            auto res = rs2_get_frame_vertices(get(), &e);
            error::handle(e);
            return (const vertex*)res;
        }
```

And as you are already here, maybe I could get a hint on how to process these vertexes.
Is it a pointer to the first vertex in an array?
How can i Iterate over it without knowing the length?

MartyG-RealSense, 2020-04-29:
Hi @PerceptMD 

Thanks for the script!  If you are asking whether the script could be added to the documentation page that you linked to, it doesn't look as though the formatting of the entries on the site allow for that.

In regard to your second question about **get_vertices**, I hope that the link below will be helpful in explaining how it operates.

https://github.com/IntelRealSense/librealsense/issues/1783

PerceptMD, 2020-04-30:
Hey @MartyG-RealSense ,

no there's a false doxygen documentation. The Function `const vertex* get_vertices(void)` returns the first Element of an array of vertices and has no input parameters. But the documentation/ doxygen comments tells this function has an input parameter.

Compare Line 742 ` * \param[in] vertex* - pointer of vertex sturcture` to Line 767 ` * \return texture_coordinate* - pointer of texture coordinates.`. 767 also describes a function with a single return param but it is marked as "return" and not as "param[in]".

Thanks, this link is really helpful!

MartyG-RealSense, 2020-04-30:
Thanks for the clarification about your concern about the documentation!

MartyG-RealSense, 2022-05-16:
Hi @PerceptMD  As the documentation change request that you made in 2020 is now 2 years old, can you confim whether you still believe that the requested change should be made, please?  Thanks!

closed 2022-05-28 by MartyG-RealSense