realsenseai/librealsense #6330: align module throwing unintelligble error (python)

cpatdowling, 2020-04-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Ubuntu 18.04.4 | 
| Kernel Version (Linux Only)    |  5.3.0-46-generic                                         | 
| Platform                                 | PC Intel Core i7-7700  |
| SDK Version                            |  liberealsense 2.34                     | 
| Language                            |  python  3.6.9 and 3.7.6                       | 
| Segment			|  other                         | 

### Issue Description

I am attempting to read a bag file into memory using the align process to match the depth and color frame FoV. Using some boilerplate code from various examples for the pyrealsense package the following function works properly if I explicitly remove declaring the aligner and using the align.process while reading frames. When I do try to use the align process, I receive the error

`Error occured during execution of the processing block! See the log for more info
16  frames read.`

First, can anyone shed any light on where I might find the log file the error is referring to? I can't seem to find any documentation. Secondly, would anyone have any idea as to what the error might be? Hoping its user error.

`def read_bag(input_file):
    frame_list = []
    try:
        config = rs.config()
        config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)
        config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)
        rs.config.enable_device_from_file(config, input_file, repeat_playback=False)
        pipeline = rs.pipeline()

        profile = pipeline.start(config)
        
        align_to = rs.stream.color
        align = rs.align(align_to)
        depth_sensor = profile.get_device().first_depth_sensor()
        depth_scale = depth_sensor.get_depth_scale()
        
        playback = profile.get_device().as_playback()
        playback.set_real_time(False)
        
        i = 0
        while True:
            frames = pipeline.wait_for_frames()
            
            playback.pause()
            frames = align.process(frames)

            color_frame = frames.get_color_frame()
            aligned_depth_frame = frames.get_depth_frame()
            
            if not color_frame or not aligned_depth_frame:
                continue
                
            ts = color_frame.get_timestamp()
            color_frame_data = np.asanyarray(color_frame.get_data())
            depth_frame_data = np.asanyarray(aligned_depth_frame.get_data())
            j = frames.get_frame_number()
            i += 1
            frame_list.append((j, ts, color_frame_data, depth_frame_data))
            playback.resume()
            
    except Exception as err:
        print(err)
    finally:
        pipeline.stop()
        print(i, " frames read.")
        return(frame_list)`

MartyG-RealSense, 2020-04-30:
Intel has a Python tutorial for reading a bag file and aligning rgb and depth if you have not seen it already.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb

I was not able to find the log location either.  I did though find a case similar to yours where a RealSense Python user was getting the same error as you when trying to use a bag file to align depth to infrared (as they did not record color in their bag).  Another RealSense user in that discussion suggested a fix.

https://github.com/IntelRealSense/librealsense/issues/5878#issuecomment-614631975

cpatdowling, 2020-04-30:
I was working off of that example, yes. A lot of my problems seem to stem from trying to sequentially read bag frames into memory for post-hoc analysis. For the moment I found a straightforward align workaround. I remove the align process from the read function altogether and define a new function that operates on a list of pyrealsense2 frame objects:

```
def align_frame_list(frame_list):
    aligned_frame_data = []
    
    #declare aligner
    align_to = rs.stream.color
    align = rs.align(align_to)
    
    for frames in frame_list:
        aligned_frames = align.process(frames)
        color_frame_data = np.asanyarray(aligned_frames.get_color_frame().get_data())
        depth_frame_data = np.asanyarray(aligned_frames.get_depth_frame().get_data())
        
        aligned_frame_data.append((color_frame_data, depth_frame_data))
        
    return(aligned_frame_data)
```
Reading frames into memory is still a brittle process for me though; the [documentation](https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0) on buffering and queueing is tricky to port to pyrealsense2.

closed 2020-04-30 by cpatdowling

MartyG-RealSense, 2020-04-30:
Glad you found a solution!

There is a Pyrealsense2 documentation page on frame queues here:

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.frame_queue.html#pyrealsense2.frame_queue