realsenseai/librealsense #6331: how to get raw IR images without turning off the laser emitter on D435i

kevin2431, 2020-04-30:
---

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435I  | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |    Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  Ubuntu 18.04                                        | 
| Platform                                 | PC  |
| SDK Version                            | liberealsense 2.34                        | 
| Language                            |  C++                     | 
| Segment			|  Robot                         | 

### Issue Description

I am working on a project which needs stereo and depth images. If turn off the emitter, I can get raw IR images without patterns. However, in this way can not get high-quality depth images.
Is there any way to get raw stereo IR images without turning off the laser emitter on D435i?
Thanks!

MartyG-RealSense, 2020-04-30:
The visibility of the dot pattern projection is related to the Laser Power setting.  So you could render the pattern virtually invisible whilst keeping the emitter active by setting laser power to '1' (a setting of '0' turns the projector off).

You can set the Laser Power in the "Controls" section of the RealSense Viewer program's options side panel, or control it with scripting:

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser

Please use the laser control code in the section headed 'Librealsense2'.

MartyG-RealSense, 2020-05-17:
Hi @kevin2431   Do you require further assistance with this question please, or can this case be closed?   Thanks!

closed 2020-05-17 by kevin2431

mydlgit, 2020-06-05:
> Hi @kevin2431 Do you require further assistance with this question please, or can this case be closed? Thanks!

> The visibility of the dot pattern projection is linked to the Laser Power setting. So you could render the pattern virtually invisible whilst keeping the emitter active by setting laser power to '1' (a setting of '0' turns the projector off).
> 
> You can set the Laser Power in the "Controls" section of the RealSense Viewer program's options side panel, or control it with scripting. The C++ version of this code is in the link below:
> 
> https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser
> 
> Please use the laser control code in the section headed 'Librealsense2

hi, how to do this in python?

MartyG-RealSense, 2020-06-05:
@mydlgit   A Python script for setting laser power is in the link below.

https://github.com/IntelRealSense/librealsense/issues/1258#issuecomment-374112555