realsenseai/librealsense #6332: Announcement: Intel white paper on open source ethernet networking for RealSense depth cameras

MartyG-RealSense, 2020-04-30:
Hi everyone,

Intel have published a new white-paper document on open-source ethernet networking for RealSense depth cameras.  It is accompanied by an article on the official RealSense blog.

**Blog article**
https://www.intelrealsense.com/depth-camera-over-ethernet-guide/

**White paper**
https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

You may also like to check out the other three new RealSense white papers published recently on the subjects of:

**Camera Self-Calibration**
**High Speed Capture Mode**
**Depth Image Compression**

https://dev.intelrealsense.com/docs/whitepapers

dorodnic, 2020-05-04:
Any questions / feedback on any of the whitepapers are welcomed

closed 2020-05-29 by RealSenseCustomerSupport

zorderjohn, 2022-06-15:
Thank you for this guide. It's really helpful. 
I recently tried to follow it and found some issues that took some time to solve:
- It should be noted the SDK version used in the prebuilt SD flash image, and that it only works with firmware version <= 5.12.7.100
- It didn't work with Raspberry PI 4 model B 8 GB (boot issues). Looks like it's built for the 4 GB model. To solve this, I replaced .bin and .elf files in boot unit for the ones that are created when installing a fresh new Raspbian distribution.

MartyG-RealSense, 2022-06-15:
Thanks very much @zorderjohn for your feedback!  The pre-built SD card image was created with SDK version 2.34.0 that dates back to March 2020.  

Yes, you are correct about the firmware version.  Firmware 5.12.7.100 should be used with SDK versions older than 2.39.0 as 2.39.0 became the recommended minimum SDK from firmware 5.12.8.200 onwards due to internal changes in the firmware.

Thanks too for sharing your method for using the SD card image with newer Pi 4 models.

SeanShi13, 2023-10-30:
@zorderjohn I also wanted to use Raspberry PI 4 model B 8 GB but encountered the same boot issues. Can you explain more about how you resolved it?  Thanks a lot.