realsenseai/librealsense #6333: matlab save and load realsense frame

EverettWang, 2020-04-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435, D435I } | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   {Win 10  |                                          | 
| SDK Version                            |   2.3.3.1.1360                          | 
| Language                            |  {malab/2018b }                          | 
| Segment			|  {others }                          | 

Hi All,

I am using D435 and D435I to do a medical related application. It requires to save a frame (we just
need a color image and its corresponding depth information). Then load it back in to matlab for further processing.  I have tried to save a frame into a mat file:

 % Make Pipeline object to manage streaming
pipe = realsense.pipeline();
% Make Colorizer object to prettify depth output
colorizer = realsense.colorizer();

% Start streaming on an arbitrary camera with default settings
profile = pipe.start();

% Get frames. We discard the 60 frames (2s) to allow
% the camera time to settle, especially color balance
for i = 1:60
    fs = pipe.wait_for_frames();
end
% 拍照的时、分、秒
date_string = datestr(datetime, 'HH_MM_SS');
%date_string = date;
% Stop streaming
pipe.stop();

% color image
color = fs.get_color_frame();
color_data = color.get_data();
color_img = permute(reshape(color_data',[3,color.get_width(),color.get_height()]),[3 2 1]);
imshow(color_img)
title(sprintf("Color image frame by %s", name));

% save it to the a file name and time
save([name date_string '.mat'], 'fs');

The file is created. But it is pretty small in size (496byte).
When I tried to load the frame into a variable, it gives me an error:

error use: librealsense_mex
null pointer passed for argument "frame_ref"

error: load_frame (line 15)
color_data = colorfs.get_data();

my loading matlab is:
% Make Pipeline object to manage streaming
pipe = realsense.pipeline();
% Make Colorizer object to prettify depth output
colorizer = realsense.colorizer();

load('Everett20_50_34.mat', 'fs');

% color image
colorfs = fs.get_color_frame();
color_data = colorfs.get_data(); <<<<< **this is line that matlab says the null pointer error**
color_img = permute(reshape(color_data',[3,colorfs.get_width(),colorfs.get_height()]),[3 2 1]);
imshow(color_img)

I am pretty sure the save and load are not working in this way. What is the correct way to save a frame and load frame? I have looked all related examples and API and didn't find any solution. Please help.

Best regards,

Everett

MartyG-RealSense, 2020-04-30:
I should emphasise from the start that I am not experienced in programming scripts for the SDK's MATLAB wrapper, so my research efforts are primarily based on comparing your code with other scripts.  So I apologise in advance for any errors in my statements.  Having said that, let's give this a try.   :)

Your code follows the structure of the **rosbag_example** sample script in the MATLAB wrapper quite closely.  After careful analysis line by line, I noticed that the main difference between your script in the sample happens once getting to this line:

**color_img = permute(reshape(color_data',[3,color.get_width(),color.get_height()]),[3 2 1]);**

In the rosbag script, **color** is a variable that has colorizer data stored in it, as defined by this line:

**color = colorizer.colorize(depth);**

[image]

In your own script though, **color** is defined as **color = fs.get_color_frame();**

This made me wonder if the color data is not getting written by the **color_img** statement because the **color.get_width(),** term in the color_img statement is expecting a **color** variable that has depth colorizer data stored in it instead of the contents of **fs.get_color_frame()**.

I hope that this description makes some kind of sense!  :)

EverettWang, 2020-05-01:
Hi Marty，

Thanks for your quick response.  I guess I am not very clear what I need. I want to save a realsense frame, the fs varaible in my last script in a .mat file so I can process it in a later time. Since the frame contains both color image and depth information, I can later load it back to do some more work (manual labeling). I just showed in the last posting a wrong way of doing it. Now I am asking the correct way of saving and loading frame file. It probably has little to do with the way to extract color information from a frame. 

Thanks for your help.

Everett

MartyG-RealSense, 2020-05-01:
I did further extensive research, knowing in advance that most of the past RealSense information that I have seen refers to loading in existing files (e.g image, ply and bag) instead of saving them within MATLAB.

I did come across a past case about saving that I handled though, where the best method available seemed to be to use the **imwrite** system of MATLAB to save an image file (**imread** is used to load a file into MATLAB).  You can use conditions in the save instruction to set up the format that the data is saved to file in.

https://uk.mathworks.com/help/matlab/ref/imwrite.html

EverettWang, 2020-05-01:
Hi Marty，

Thanks again for your quick suggestions. But saving image only solves half of my problem. The frame contains both color image (which can be saved by imwrite) and depth information, which I want to use to construct point cloud. 

I also looked at your suggestions of saving information into bag file. It will contain both depth and color image information. The drawback is that it will contain many more frames, occupying more space. Since it seems not easy to save frame, let me try using bag file for my application.

Best regards,

Everett

closed 2020-05-01 by EverettWang