realsenseai/librealsense #6334: Cannot stream depth, color, infrared at once using pyrealsense2

StaringWhere, 2020-05-01:
| Required Info               |             |
| --------------------------- | ----------- |
| Camera Model                | D400        |
| Firmware Version            | 05.12.03.00 |
| Operating System & Version  | Ubuntu18.04 |
| Kernel Version (Linux Only) | 5.3.0-47    |
| Platform                    | PC          |
| SDK Version                 | 2.34.0      |
| Language                    | python      |
| Segment                     | others      |

### Issue Description
I was trying to gather the depth, color, infrared data using D415, but there was a lot of frame drops in color and infrared streaming, and depth streaming was freezed at the first frame. Frame drops stopped if I only stream two of them. Also It works perfectly in realsense viewer (stream depth, color, infrared at the same time).

Here is my code

```python
import pyrealsense2 as rs
import numpy as np
import cv2
from time import time

# Create a pipeline
pipeline = rs.pipeline()

# Create a config and configure the pipeline to stream
#  different resolutions of color and depth streams
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
config.enable_stream(rs.stream.infrared, 640, 480, rs.format.bgr8, 30)

# Start streaming
profile = pipeline.start(config)

try:
    while True:
        # Get frameset of color and depth
        start = time()
        frames = pipeline.wait_for_frames()
        end = time()
        print(end-start)

        # Get aligned frames
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        infrared_frame = frames.get_infrared_frame()

        # Validate that both frames are valid
        # if not depth_frame or not color_frame or not infrared_frame:
        # if not aligned_depth_frame or not color_frame:
            # continue

        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        infrared_image = np.asanyarray(infrared_frame.get_data())

        # Render images
        colorizer = rs.colorizer() 
        depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())
        images = np.hstack((color_image, depth_colormap, infrared_image))
        # images = np.hstack((bg_removed, depth_colormap))
        cv2.namedWindow('Align Example', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('Align Example', images)
        key = cv2.waitKey(1)
        # Press esc or 'q' to close the image window
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
finally:
    pipeline.stop()
```

Time used by `pipeline.wait_for_frames()`

```
0.01884007453918457
0.022507667541503906
0.018776416778564453
0.02047276496887207
0.00011014938354492188
0.050071001052856445
0.1531381607055664
1.8667693138122559
7.605552673339844e-05
0.11472535133361816
7.43865966796875e-05
0.1430492401123047
7.390975952148438e-05
0.13817524909973145
7.82012939453125e-05
1.487736463546753
0.00018286705017089844
0.11703133583068848
8.606910705566406e-05
0.1409614086151123
0.1532881259918213
1.4963865280151367
8.416175842285156e-05
0.11475396156311035
7.295608520507812e-05
0.14925622940063477
7.343292236328125e-05
0.13791990280151367
1.4948937892913818
8.916854858398438e-05
0.10437273979187012
1.8357906341552734
0.12919402122497559
```

MartyG-RealSense, 2020-05-01:
If you are aiming to stream infrared, could you try changing the format of the infrared stream from bgr8 to **y8** please?  bgr8 would be more likely to be used if trying to stream a color image from the IR sensor instead of an infrared image.

StaringWhere, 2020-05-01:
Hi Marty,
Thank you for your quick responce! I've changed bgr8 to y8, there is less frame drops now, but depth frame still freezes at the first frame.

MartyG-RealSense, 2020-05-01:
You are very welcome   :)

Could you describe the intended output of your application please?  Does the application window contain three separate streams (RGB, depth and infrared), or are you trying to overlay all three streams in a single image?  Thanks!

StaringWhere, 2020-05-01:
Thank you so much! Changing FPS from 30 to 15 solved the problem.

MartyG-RealSense, 2020-05-01:
I'm very glad that you found a solution!

A recommended way to improve performance is to build the Librealsense SDK in 'Release' mode instead of 'Debug' mode if you are not doing so already.

[image]

This can be done in the SDK's CMake build instruction by including this term:

**-DCMAKE_BUILD_TYPE=release**

Another method of improving performance is to offload some of the CPU processing work onto the graphics GPU, if your computer has a good GPU component.  This can be done by building the SDK with CUDA support if you have an Nvidia type GPU in your computer, or by using "GLSL Processing Blocks" if you have any other type of GPU.

https://github.com/IntelRealSense/librealsense/issues/4905#issuecomment-533854888

StaringWhere, 2020-05-01:
Thank you for the good advice, I will try those👍

closed 2020-05-04 by StaringWhere