realsenseai/librealsense #6336: Border around depth image

droidXrobot, 2020-05-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.0  | 
| Language                            |  C# | 
| Segment			| Other  | 

### Issue Description
When using a depth cutoff in Unity, and the Realsense viewer, there is a border around one side of the depth and color point cloud. Is there any way to ensure the background pixels doesn't seep into the image?

The images below show depth cutoff in Unity and a normal colored pointcloud in the Realsense viewer. In Realsense it looks like the pointcloud and colored image aren't aligned. Could it be a calibration issue?

Unity
[image]

Realsense viewer
[image]

MartyG-RealSense, 2020-05-01:
In the RealSense Viewer image, the black silhouette around the body is an effect known as "occlusion" that is generated because the RGB and the IR sensors are in different physical positions on the camera.  

The RealSense Viewer has an 'Occlusion Removal' option that you could try.

https://github.com/IntelRealSense/librealsense/issues/2445#issuecomment-424481687

To use this option, please follow the steps below.

**STEP 1**

Enable 3D point cloud mode on the RealSense Viewer.

**STEP 2**

Enable both the **Depth** and **RGB** streams.

[image]

[image]

**STEP 3**

Left-click on the 'Texture Source' drop-down menu at the top of the Viewer window and set it to use Color as the texture source instead of Depth.

[image]

**STEP 4**

A drop-down "Occlusion removal" menu will appear at the top of the screen.  Left-click on it and select the 'Heuristic' option from the menu.

[image]

droidXrobot, 2020-05-01:
> #2445 (comment)

Thanks Marty,  it also seems the the colored image is shifted in slightly
to one side. Is there a way to fix that shift?

Sent from my mobile. Please pardon the brevity.

MartyG-RealSense, 2020-05-01:
In regard to the face edge being cut off at the side in the first image, if that was what you were asking about when you referenced the color image, you may find the discussion in the link below useful.

https://github.com/IntelRealSense/librealsense/issues/4629

dorodnic, 2020-05-05:
Also, occlusion invalidation is going to be enabled by default starting with next release. 
I agree with @MartyG-RealSense that what you describe as shift is due to occlusion.

MartyG-RealSense, 2020-05-17:
Hi @droidXrobot   Do you require further assistance with this question please, or is it okay to close this case?   Thanks!

closed 2020-05-24 by MartyG-RealSense